project-chip/connectedhomeip #1425: Missing various defines/declarations in SiLabs code contribution

bzbarsky-apple, 2020-07-02:
#### Problem
I am trying to compile against some of the headers from https://github.com/project-chip/connectedhomeip/pull/1164 and running into a few missing definitions:

* [ ] `EmberIncomingMessageType`
* [ ] `EmberOutgoingMessageType`
* [ ] `EUI64_SIZE`
* [ ] `EMBER_ENCRYPTION_KEY_SIZE`
* [ ] `SIGNED_ENUM`
* [ ] `EmberEventControl`
* [ ] `EmberStatus`
* [ ] `EmberAfMessageSentFunction`

 #### Proposed Solution
Figure out which of these should be defined in the ZCL code (e.g. in `types_stub.h`), which should be defined in CHIP somehow, and which we just don't need (e.g. because we can remove the parts of the SiLabs code/headers that use them).  then take appropriate action.

@bhaskar-apple @selissia

rwalker-apple, 2020-07-04:
I'm also looking for:

- [ ]  `EmberApsFrame`

bhaskar-apple, 2020-07-06:
@bzbarsky-apple - some of these are to do with security / networking e.g. `EmberInitialSecurityState` 
We may not need them as we have our own layer. And might be safe to remove any code that uses it.

bzbarsky-apple, 2020-07-06:
>  EmberApsFrame

@rwalker-apple It's in https://github.com/project-chip/connectedhomeip/pull/1164 (added after you grabbed your copy of it).  See https://github.com/project-chip/connectedhomeip/pull/1164/commits/113fac2de64556ad1a4731428d88c88f747b1e84

bzbarsky-apple, 2020-07-06:
> We may not need them as we have our own layer. And might be safe to remove any code that uses it.

That seems plausible.  For now I just listed the things that were missing so we can evaluate each one and decide whether we need its definition from SiLabs or not and track that happening.

bhaskar-apple, 2020-07-06:
> > We may not need them as we have our own layer. And might be safe to remove any code that uses it.
> 
> That seems plausible. For now I just listed the things that were missing so we can evaluate each one and decide whether we need its definition from SiLabs or not and track that happening.

That's fair. Let's get these all in then! :)

bzbarsky-apple, 2020-07-06:
To be clear, what I'd really like here is for @wehale or @selissia or @tecimovic to go through the list and give us some guidance on which of these we really are likely to need and how we should get them vs which ones we should just remove the code that uses them or change it to use CHIP types.

wehale, 2020-07-07:
Most of these are specifically related to ZigbeePro so I don't think you want them in CHIP, you want some CHIP equivalent. We should go through them, but I can see that, since they are not included in the open source they come from the stack which is a ZigbeePro Stack.

wehale, 2020-07-07:
Lets plan to go through these with @tecimovic, @selissia, for each I expect you need to know, where it comes from, what it is used for and if it is relevant to CHIP. This is exactly the crusty portion of the interface we're going to run into where some CHIP equivalents will need to be provided in order to bolt this onto CHIP

selissia, 2020-07-07:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I plan to go through this in the next couple of days.

bzbarsky-apple, 2020-07-07:
That sounds lovely.  I don't know that we need a sync meeting to go through these; async short text blurbs would be just fine on my end.

selissia, 2020-07-09:
[types_stub.zip](https://github.com/project-chip/connectedhomeip/files/4899449/types_stub.zip)
Attaching updated types_stub.c here as the commits to my private fork are not propagated to the PR.

selissia, 2020-07-09:
Added all missing definitions in the attachment above. 
Many of these types and functions that use them are not needed and should be purged from the code. Some of these are:
EmberZigbeePacketType
EmberZigbeeNetwork
SleepModes

bzbarsky-apple, 2020-07-13:
@selissia Thank you for that zip file!

I grabbed the header from it, with the following modifications:

1. `EMBER_MIN_802_15_4_CHANNEL_NUMBER` should be 11, not 1.  Probably doesn't matter for CHIP anyway.
2. `EmberNetworkStatus` was in the header twice; I just removed the second declaration.

It's still missing `EMBER_BROADCAST_ENDPOINT`, `MEMSET`, `MEMCOPY`, `MEMMOVE`, `MEMCOMPARE`, `BIT`, `LOW_BYTE`, `HIGH_BYTE`, `BYTE_0`, `BYTE_1`, `BYTE_2`, `BYTE_3`, `MAIN_FUNCTION_PARAMETERS`.  Some of those I had not managed to get on the list, apparently; some might have gotten missed while going through the list.

selissia, 2020-07-13:
Attached a file with missing definitions. Looks like `EmberBindingType` might actually be needed, the rest I think can be made into `uint8_t` , I included them anyway.

[missing_types.zip](https://github.com/project-chip/connectedhomeip/files/4914820/missing_types.zip)

bzbarsky-apple, 2020-07-14:
Thanks, that is very helpful.  I assume you're OK with me just fixing the 6 items from <https://github.com/project-chip/connectedhomeip/issues/1425#issuecomment-657740924>, in terms of licensing?

selissia, 2020-07-14:
> I assume you're OK with me just fixing the 6 items from #1425 (comment), in terms of licensing?
Yes, absolutely.

bzbarsky-apple, 2020-07-14:
One other compile issue I ran into: `types_stub.h` needs to include `string.h` for the `mem*` functions.  With that fixed, things are good now.

closed 2020-07-14 by bzbarsky-apple