project-chip/connectedhomeip #1435: VScode setup failing

markus-becker-tridonic-com, 2020-07-03:
#### Problem

When following https://github.com/project-chip/connectedhomeip/blob/master/docs/VSCODE_DEVELOPMENT.md "Bootstrapping your source tree (one time)" fails with the following error message:

```
> Executing task: scripts/build/bootstrap.sh <

+ env
LANG=en_US.UTF-8
HOSTNAME=50eb66b84a08
AMD_ENTRYPOINT=vs/server/remoteExtensionHostProcess
COLORTERM=truecolor
SSH_AUTH_SOCK=/tmp/vscode-ssh-auth-7f747b26d9f1e7289292f75047a911dc946c9f5a.sock
IDF_TOOLS_PATH=/var/.espressif
APPLICATION_INSIGHTS_NO_DIAGNOSTIC_CHANNEL=true
PWD=/workspaces/connectedhomeip
HOME=/home/vscode
VSCODE_GIT_ASKPASS_NODE=/home/vscode/.vscode-server/bin/cd9ea6488829f560dc949a8b2fb789f3cdc05f5d/node
GIT_PS1_SHOWDIRTYSTATE=1
TERM_PROGRAM=vscode
TERM_PROGRAM_VERSION=1.46.1
GIT_PS1_SHOWCOLORHINTS=true
REMOTE_CONTAINERS_IPC=/tmp/vscode-remote-containers-ipc-7f747b26d9f1e7289292f75047a911dc946c9f5a.sock
VSCODE_IPC_HOOK_CLI=/tmp/vscode-ipc-21de703f-3b0d-4cfd-847b-663252128669.sock
QEMU_ESP32=/var/qemu_esp32/xtensa-softmmu/qemu-system-xtensa
VSCODE_GIT_ASKPASS_MAIN=/home/vscode/.vscode-server/bin/cd9ea6488829f560dc949a8b2fb789f3cdc05f5d/extensions/git/dist/askpass-main.js
QEMU_ESP32_DIR=/var/qemu_esp32
SHELL=/bin/bash
TERM=xterm-256color
GIT_PS1_SHOWSTASHSTATE=1
SHLVL=2
IDF_PATH=/var/esp-idf/
VSCODE_GIT_IPC_HANDLE=/tmp/vscode-git-080ee6a1af.sock
PIPE_LOGGING=true
GIT_ASKPASS=/home/vscode/.vscode-server/bin/cd9ea6488829f560dc949a8b2fb789f3cdc05f5d/extensions/git/dist/askpass.sh
PATH=/home/vscode/.vscode-server/bin/cd9ea6488829f560dc949a8b2fb789f3cdc05f5d/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
VERBOSE_LOGGING=true
_=/usr/bin/env
+ [[ ! -f build/default/config.status ]]
+ mkdir -p build/default
+ cd build/default
+ ../../bootstrap-configure -C --enable-debug --enable-coverage
../../bootstrap-configure: line 1: third_party/nlbuild-autotools/repo/scripts/bootstrap-configure: No such file or directory
The terminal process terminated with exit code: 127

Terminal will be reused by tasks, press any key to close it.
```

The file ${CHIP_ROOT}/third_party/nlbuild-autotools/repo/scripts/bootstrap-configure is there:
```
~/src/chip/connectedhomeip (master)
$ ls -la third_party/nlbuild-autotools/repo/scripts/bootstrap-configure
-rwxr-xr-x 1 Markus.Becker 1049089 1410 Jul  3 09:14 third_party/nlbuild-autotools/repo/scripts/bootstrap-configure*
```

 * system configuration
Windows 10, VScode 1.46.1, git SHA 9c0a67d66c0aaa041f5e6b752cc3383e96e2ed76

woody-apple, 2020-07-23:
Is this still failing for you?

woody-apple, 2020-07-23:
@markus-becker-tridonic-com ?

markus-becker-tridonic-com, 2020-07-26:
Switched to a Mac there it is working. Need to check on Windows once again.

markus-becker-tridonic-com, 2020-09-11:
Tried once more on a Windows 10 machine. Now with the GN based VSCode tasks.

```
> Executing task: bash scripts/bootstrap.sh <

Submodule 'qrcode' (https://github.com/nayuki/QR-Code-generator.git) registered for path 'examples/common/QRCode/repo'
Submodule 'm5stack-tft' (https://github.com/jeremyjh/ESP32_TFT_library.git) registered for path 'examples/common/m5stack-tft/repo'
Submodule 'bluez' (git://git.kernel.org/pub/scm/bluetooth/bluez.git) registered for path 'third_party/bluez/repo'
Submodule 'cirque' (https://github.com/openweave/cirque.git) registered for path 'third_party/cirque/repo'
Submodule 'mbedtls' (https://github.com/ARMmbed/mbedtls.git) registered for path 'third_party/mbedtls/repo'
Submodule 'nlassert' (https://github.com/nestlabs/nlassert.git) registered for path 'third_party/nlassert/repo'
Submodule 'nlfaultinjection' (https://github.com/nestlabs/nlfaultinjection.git) registered for path 'third_party/nlfaultinjection/repo'
Submodule 'nlio' (https://github.com/nestlabs/nlio.git) registered for path 'third_party/nlio/repo'
Submodule 'nlunit-test' (https://github.com/nestlabs/nlunit-test.git) registered for path 'third_party/nlunit-test/repo'
Submodule 'openthread' (https://github.com/openthread/openthread.git) registered for path 'third_party/openthread/repo'
Submodule 'ot-br-posix' (https://github.com/openthread/ot-br-posix.git) registered for path 'third_party/ot-br-posix/repo'
Submodule 'ot-commissioner' (https://github.com/openthread/ot-commissioner) registered for path 'third_party/ot-commissioner/repo'
Submodule 'pigweed' (https://pigweed.googlesource.com/pigweed/pigweed) registered for path 'third_party/pigweed/repo'
Cloning into '/workspaces/connectedhomeip/examples/common/QRCode/repo'...
Cloning into '/workspaces/connectedhomeip/examples/common/m5stack-tft/repo'...
Cloning into '/workspaces/connectedhomeip/third_party/bluez/repo'...
Cloning into '/workspaces/connectedhomeip/third_party/cirque/repo'...
Cloning into '/workspaces/connectedhomeip/third_party/mbedtls/repo'...
Cloning into '/workspaces/connectedhomeip/third_party/nlassert/repo'...
Cloning into '/workspaces/connectedhomeip/third_party/nlfaultinjection/repo'...
Cloning into '/workspaces/connectedhomeip/third_party/nlio/repo'...
Cloning into '/workspaces/connectedhomeip/third_party/nlunit-test/repo'...
Cloning into '/workspaces/connectedhomeip/third_party/openthread/repo'...
Cloning into '/workspaces/connectedhomeip/third_party/ot-br-posix/repo'...
Cloning into '/workspaces/connectedhomeip/third_party/ot-commissioner/repo'...
Cloning into '/workspaces/connectedhomeip/third_party/pigweed/repo'...
Submodule path 'examples/common/QRCode/repo': checked out '08ac806145aa6ab5d3e9df61d023092bb6cfe761'
Submodule path 'examples/common/m5stack-tft/repo': checked out 'a1ece2c500cf179db8f0c7dbc7bb8f3ace562142'
Submodule path 'third_party/bluez/repo': checked out '179ccb9362b7eb2d56412acdf9a281ba495f75b6'
Submodule path 'third_party/cirque/repo': checked out 'bc42efe2e0e8f4e0eed7bd08a51e046943a11b98'
Submodule path 'third_party/mbedtls/repo': checked out 'ca933c7e0c9e84738b168b6b0feb89af4183a60a'
Submodule path 'third_party/nlassert/repo': checked out 'bd2f082102d4456ca750189ee853fa8c7443af59'
Submodule path 'third_party/nlfaultinjection/repo': checked out '79f92f309de58c4591f53a9b492600550b473c84'
Submodule path 'third_party/nlio/repo': checked out '8a4048faf7c84c6c9e3c9b3e79f7dc48d5d69f71'
Submodule path 'third_party/nlunit-test/repo': checked out 'ef5a2b9a8b71ad592fcf3681968fd63c844e3d81'
Submodule path 'third_party/openthread/repo': checked out 'c9004a6f3f129778fcab7dfcf9060f0f168c06b8'
Submodule path 'third_party/ot-br-posix/repo': checked out 'ea9cd69aa2eed3f320f6a80a4d0b966959c19ff0'
Submodule path 'third_party/ot-commissioner/repo': checked out 'ad96e6f6f5b1915844ba4481bf8d0b66af91175e'
Submodule path 'third_party/pigweed/repo': checked out '133eb7dc55505fde3212a61a333cd2e83acfbac6'

  WELCOME TO...

  ▄███▒  ░▓█  ░▓█ ░▓█▓ ▒█████▄
 ██▒ ▀█▒  ▒█   ▒█  ░█▒  ▒█░  █░
 █▓░      ▒██████  ░█▒  ▒█▄▄▄█░
 ▓█   █▒  ▒█   ▒█  ░█░  ▒█▀
 ░▓███▀  ░▓███░▓█▒ ░█░  ▒█

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done
  Setting up Python environment.....done
  Setting up Host tools.............done

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Python environment.....done
  Setting environment variables for Host tools.............done

Sanity checking the environment:

20200911 13:07:52 INF Environment passes all checks!
[... 97 lines cut ...]
_=/usr/bin/env
+ gn gen --root=examples/shell out/shell --args=
ERROR at //args.gni:15:1: Unable to load "/workspaces/connectedhomeip/examples/shell/build_overrides/chip.gni".
import("//build_overrides/chip.gni")
^----------------------------------
See //.gn:25:3: whence it was imported.
  import("//args.gni")
  ^------------------
The terminal process "/bin/bash '-c', 'scripts/examples/gn_build_example.sh examples/shell out/shell'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

The documentation mentions WSL and Docker:
`Windows 10 Pro + WSL + Ubuntu 18 LTS`
But it seems only Docker is used later on:
```
Install the Remote - Containers extension for Visual Studio Code, this extension allows you to use docker containers as a development backend.
[....]
Now your local machine is building a docker image that has all the tools necessary to build and test CHIP. This can take some time, but will eventually complete and open up the source tree
```

Is the cloning/bootstrapping supposed to happen in WSL/Ubuntu?

SHA: b31d453e1a44f1298896085f8dcd87c43e2e5735

sweetymhaiske, 2021-07-06:
@markus-becker-tridonic-com Are you still facing the issue?

turon, 2022-01-31:
This issue is old and may be closable, but only once there is validation that v1 SDK works properly on Windows.
Removing `v1_triage_split_4`.

closed 2022-10-09 by stale[bot]