project-chip/connectedhomeip #1439: [INSERT EMAIL ADDRESS] placeholder in Code of Conduct

coderbyheart, 2020-07-03:
I applaud the fact that this project decided to adopt a Code of Conduct!

However, the [current version](https://github.com/project-chip/connectedhomeip/blob/a52635d5d5577f99182235ebbdcc1b5ddcc02b03/CODE_OF_CONDUCT.md) has a blank:

> Instances of abusive, harassing, or otherwise unacceptable behavior may be reported by contacting the Project CHIP open source team at **[INSERT EMAIL ADDRESS]**.

Please provide a way to contact the enforcement team of Project CHIP in a way that protects the identity of the person reaching out.

woody-apple, 2020-07-23:
This was resolved, sorry for the delay!

closed 2020-07-23 by woody-apple

coderbyheart, 2020-07-26:
@woody-apple this is not resolved: https://github.com/project-chip/connectedhomeip/blob/8eb18b8aec64f85ed55079619974b2e030105575/CODE_OF_CONDUCT.md still has `[INSERT EMAIL ADDRESS]`