project-chip/connectedhomeip #1444: examples print unsuppressable #pragma messages during the build

mspang, 2020-07-06:
Ideally the examples should be warnings-clean. Most compiler warnings can be disabled via cflags, but ones generated from the preprocessor cannot. 

/home/spang/connectedhomeip2/src/inet/../../src/inet/IPEndPointBasis.cpp: In member function 'INET_ERROR chip::Inet::IPEndPointBasis::JoinMulticastGroup(chip::Inet::InterfaceId, const chip::Inet::IPAddress&)':
/home/spang/connectedhomeip2/src/inet/../../src/inet/IPEndPointBasis.cpp:460:17: note: #pragma message: 
 Please enable LWIP_IPV6_MLD && LWIP_IPV6_ND && LWIP_IPV6 for IPv6 JoinMulticastGroup and LeaveMulticastGroup support.
  460 | #pragma message "\n \
      |                 ^~~~~
  461 | Please enable LWIP_IPV6_MLD && LWIP_IPV6_ND && LWIP_IPV6 for IPv6 JoinMulticastGroup and LeaveMulticastGroup support."
      | ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/spang/connectedhomeip2/src/inet/../../src/inet/IPEndPointBasis.cpp: In member function 'INET_ERROR chip::Inet::IPEndPointBasis::LeaveMulticastGroup(chip::Inet::InterfaceId, const chip::Inet::IPAddress&)':
/home/spang/connectedhomeip2/src/inet/../../src/inet/IPEndPointBasis.cpp:549:17: note: #pragma message: 
 Please enable LWIP_IPV6_MLD && LWIP_IPV6_ND && LWIP_IPV6 for IPv6 JoinMulticastGroup and LeaveMulticastGroup support.
  549 | #pragma message "\n \
      |                 ^~~~~
  550 | Please enable LWIP_IPV6_MLD && LWIP_IPV6_ND && LWIP_IPV6 for IPv6 JoinMulticastGroup and LeaveMulticastGroup support."
      | ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

gerickson, 2020-07-06:
Any suggestions for alternatives beyond flat out removal?

mspang, 2020-07-06:
Sent some ideas on the PR:

1) The description suggests one: enable these in the examples (and probably turn it into an #warning in that case if we really don't intend to allow this combination).

2) If we intend to allow it and still want to give a hint we could handle it via documentation, or alternatively 3) add another #define like CHIP_SUPPRESS_CONFIG_RECOMMENDATIONS so there's an opt out. Currently it is impossible to make the examples build clean of diagnostic messages and the goal is mainly to make it possible.

PR #1445 (2020-07-07): Remove uses of "pragma message" for unsuppressable warnings
Compiling this file generates unsuppressable output when certain lwIP
options are disabled. If these configurations are legal, then merely
compiling shouldn't print out an objection to them. If they are truly
not intended to be used, the lwIP configs for the examples should be
corrected.

Fixes #1444
commit 1a2aca63 (2020-07-07): Remove uses of "pragma message" for unsuppressable warnings (#1445)

Compiling IPEndPointBasis.cpp generates unsuppressable output when
certain lwIP options are disabled. If disabling these options is legal,
then merely compiling shouldn't print out an objection to them. I
disabling them is not legal, then the lwIP configs for the examples
should be corrected.

Fixes #1444
closed 2020-07-07 by woody-apple