project-chip/connectedhomeip #1486: Make doc does not work

andy31415, 2020-07-07:
#### Problem
`make help` claims that doc should work .Attempting it results in failure.

 #### Proposed Solution
Figure out if we should have a doc step and if yes, then make it actually work

andy31415, 2020-07-07:
@gerickson  found that `make -C docs` is the magic incantation.

`make help` should be updated then

andy31415, 2020-07-07:
* the -C docs did something, however I still cannot see the output. Still needs investigation

vivien-apple, 2020-07-07:
For info here is how I can successfully build docs from the vscode container on top of a clean tree:
```
 $ sudo apt-get install doxygen 
 $ mkdir build/documentation
 $ cd build/documentation
 $ ../../bootstrap-configure --enable-docs
 $ make docdist
```

Outputs should be in `build/documentations/docs/`

Beside the missing doxygen into the Dockerfile, not sure what else I need to fix besides documentation.

vivien-apple, 2020-07-08:
Simpler:

```
$ sudo apt-get install doxygen
$ DOCS=1 make -f Makefile-Standalone
```

Outputs should be in `build/target_name/docs/html/`

vivien-apple, 2020-07-08:
@andy31415 I added a PR with some doc.
I would like to know if the last steps works for you, and if so if the documentation PR is good enough.

PR #1496 (2020-07-09): Add some documentation about generating the doxygen documentation
 #### Problem

People have hard time finding how to generate the documentation.

 #### Summary of Changes
 * Add some documentation in building about generating the documentation.

fixes #1486
closed 2020-07-09 by woody-apple