project-chip/connectedhomeip #1488: android-ndk-which missing

markus-becker-tridonic-com, 2020-07-07:
#### Problem

While following
https://github.com/project-chip/connectedhomeip/blob/master/docs/BUILDING.md#build-android
at `make -f Makefile-Android` a script `android-ndk-which` (https://github.com/project-chip/connectedhomeip/blob/master/Makefile-Android#L69) is referenced. It is not part of the repository.

wgtdkp, 2020-07-19:
I ran into the same issue, seems the Android build is broken.

@turon

wgtdkp, 2020-07-19:
@vidhis88 I would expect the android demo requires a prebuilt chip library(.jar and .so). Can you share how you build the demo while the Android build is broken?

vidhis88, 2020-07-22:
@wgtdkp I've updated the steps to build CHIP for Android in the README in #1662.
I will update the Makefile-Android eventually but it needs more things to be figured out as that was based on ndk r10 and we are now on r19. 
In the meantime, the README should help.

saurabhst, 2020-07-22:
@vidhis88 any reason of not using the latest ndk stable version `r21` at https://developer.android.com/ndk/downloads ?

vidhis88, 2020-07-22:
> @vidhis88 any reason of not using the latest ndk stable version `r21` at https://developer.android.com/ndk/downloads ?

You're right, I am also locally on r21. I mistakenly had r19 in mind. Basically, ndk  r14 onwards, the ndk changed significantly. The instructions in the README will work for all versions above 14.

Mainly want to call out that the Makefile-Android is incompatible with the newer ndk versions.

vidhis88, 2020-08-05:
Should this issue be closed since the Android README contains up to date steps to build CHIP for Android apps?

saurabhst, 2020-08-05:
Yup 👍 from my side . Havnt heard any recent build breaks from any one on this

markus-becker-tridonic-com, 2020-08-06:
Yep, build works for me as well.

Some additional notes that could improve the README:
* Explicitly state that NDK version 21.0.6113669 is required.
* On linux-x86_64 I needed to adapt the TARGET variables:
```
export OS_VARIANT=linux-x86_64

export TARGET1=arm-linux-androideabi
export TARGET2=armv7a-linux-androideabi

export ANDROID_NDK_HOME=/home/markus/Android/Sdk/ndk/
export TOOLCHAIN=$ANDROID_NDK_HOME/toolchains/llvm/prebuilt/$OS_VARIANT

export JAVA_HOME=/usr/lib/jvm/java-11-openjdk-amd64/bin

export API=21
export AR=$TOOLCHAIN/bin/$TARGET1-ar
export AS=$TOOLCHAIN/bin/$TARGET1-as
export CC=$TOOLCHAIN/bin/$TARGET2$API-clang
export CXX=$TOOLCHAIN/bin/$TARGET2$API-clang++
export LD=$TOOLCHAIN/bin/$TARGET1-ld
export RANLIB=$TOOLCHAIN/bin/$TARGET1-ranlib
export STRIP=$TOOLCHAIN/bin/$TARGET1-strip

export JNI_INCLUDE_DIRS=$ANDROID_NDK_HOME/sysroot/usr/include
```
and then use `$TARGET1` at configure step `../connectedhomeip/configure --host=$TARGET1 --with-crypto=mbedtls --enable-tests=no --enable-shared --with-device-layer=none`.

* Also slightly improve by explaining which jars and so files to copy where:
```
cp ./src/controller/java/CHIPController.jar ./src/setup_payload/java/SetupPayloadParser.jar ../connectedhomeip/src/android/CHIPTool/app/libs/

cp ./src/controller/java/.libs/libCHIPController.so
./src/setup_payload/java/.libs/libSetupPayloadParser.so ../connectedhomeip/src/android/CHIPTool/app/src/main/jniLibs/
```

closed 2020-08-06 by markus-becker-tridonic-com