project-chip/connectedhomeip #1505: pigweed bootstrap takes a long time inCI

rwalker-apple, 2020-07-08:
#### Problem
pigweed is designed to work really well in local dev environments.  as a result, during bootstrap it binds itself to the client repository by setting up a virtual environment in the dev tree and binding environment variables to the tree's location

in our current CI system, the dev tree starts from a clean slate, so every CI run needs to bootstrap pigweed (which is relatively expensive)

 #### Proposed Solution
research how best to work with pigweed in CI

keir, 2020-07-08:
We've started investigating ways to improve the bootstrap time. One solution is to have a Docker image that has the environment pre-setup; then the build can run from that.

**Docker pros**
* Can leverage the infrastructure around Docker, including integration by CI providers
* Will work quickly in CI thanks to the various CI system's Docker caching
* Will work quickly locally thanks to Docker's built-in caching on your machine
* Can work in parallel with a host-based build that uses the bootstrap directly (no conflicts)

**Docker cons**
* Doesn't help with Windows or Mac host-based CI, since Docker containers are essentially Linux VMs.

**Docker implementation strategies**
There are a couple ways the Docker workflow could be implemented:
* CHIP could have its own image that directly stands up the environment (either using bootstrap or manually), which is updated by CHIP.
*  There could be an upstream frequently-updated Docker image (perhaps pigweed/pigweed:master), then CHIP could derive from that image in the `FROM` line of the CHIP Dockerfile. Then whenever the Pigweed submodule in CHIP is bumped, the CHIP image would also have to be re-built.

In both implementations, the resulting Docker images can be used both by upstream and downstream CI. Local development could also be done with the image, though we feel most people will have a better experience with the local-native setup.

**Other approaches**
Other solutions require some sort of custom local caching, which is harder since each CI environment is different. Rob, can you offer any more details about your CI setup? Would a Docker-based solution work? Or is there a way to have persistent builders in your CI system?

rwalker-apple, 2020-07-09:
CI setup: we use github actions with custom docker images

Those docker images make no assumptions about where the source tree will land, which I think is a requirement of pigweed?  maybe pigweed initialization can be run separately from binding to a source tree?  if so, we can put pigweed in our docker images and bind to pigweed after checkout in CI

keir, 2020-07-09:
There is no fundamental reason the bootstrap needs to be tied to the source directory; it's just that in our current setup that's what is done. 

We will spend some time tomorrow making a documented & supported approach for splitting the checkout from the environment so that it is easy to dockerize the environment without the source; stay tuned.

mspang, 2020-07-09:
Based on the githubs docs, it's possible to have persistence if you host the runners yourself. There are some other overheads that would be probably be reduced in that case, but we'd have to stomach the costs and risks of managing it.

rwalker-apple, 2020-07-09:
> Based on the githubs docs, it's possible to have persistence if you host the runners yourself. There are some other overheads that would be reduced in that case, but we'd have to stomach the costs and risks of managing it.

If we introduce something that requires building+hosting runners (like tethered HW, e.g.), yeah: we could take that step.  I'd prefer for us to stay "ops free" as long as possible, though.

keir, 2020-07-09:
We have started the changes to split off the environment to support the Docker workflow:
https://pigweed-review.git.corp.google.com/c/pigweed/pigweed/+/13600

A couple more patches will be needed.

keir, 2020-07-13:
We've continued to work on this, but ran into some Docker issues. We will have an update and hopefully a working Docker solution this week.

rwalker-apple, 2020-07-14:
@keir does pigweed have a public issue tracking system?  I'm finding some things in activate.sh that look a bit off...

mspang, 2020-07-14:
https://bugs.chromium.org/p/pigweed/issues/list?q=&can=2

rwalker-apple, 2020-07-14:
> https://bugs.chromium.org/p/pigweed/issues/list?q=&can=2
[image]

keir, 2020-07-21:
Sorry about the access issue Robb! Pigweed's issue tracker is still closed due to some internal red tape, but we plan to open it this quarter.

We are continuing to work on Docker integration:
https://pigweed-review.googlesource.com/q/docker

We have pre-caching the CIPD loaded dependencies working, but it needs some more refinement before we suggest others copy our approach.

keir, 2020-07-23:
We have now opened Pigweed's issue tracker:

https://bugs.chromium.org/p/pigweed/issues/list

We are still working on the Docker solution. There is an upstream docker file in place, but we haven't done the documentation and last mile work. We should have an update next week.

keir, 2020-08-03:
We are still working on this; you can see progress on this bug:
https://bugs.chromium.org/p/pigweed/issues/detail?id=216

keir, 2020-08-04:
We got our staged container builds working today:
https://ci.chromium.org/p/pigweed/builders/try/pigweed-linux-container/b8872946964556317200

Once we have our docs together, we may reach out to have other teams try either the containers we've created, or replicating the contents of our Dockerfile.

andy31415, 2021-06-17:
I can currently see pigweed having a separate dockerfile that sets us cipd and python: https://github.com/google/pigweed/blob/main/docker/Dockerfile.cache

We could replicate this inside docker, however concerns are:
  - will this really speed up (CIPD brings in GB of compiler bits, will placing these in the image be significantly faster than downloading?)
  - Python and even CIPD dependencies are based on python requirements and pigweed update. Manual docker updates to keep things in sync seems more likely to be errorprone

closed 2022-01-25 by woody-apple