project-chip/connectedhomeip #1513: GN build parity: packaging/install

mspang, 2020-07-08:
GN build needs support for installation.

1) Libraries should be able to declare their public headers as metadata 
2) This metadata must propagate through dependencies
3) Implement an action that packages up static libs & their headers

mspang, 2020-08-19:
I am unsure if this is blocking. The integrations I'm involved build from source and don't make extra copies of artifacts or sources along the way.

caipiblack, 2021-10-05:
Hi,

I'm also looking for this feature, to deploy the library and headers automatically with a GN rule instead of doing it "by hand" (or external script)

Could someone treat this issue please ?

Is there any blocking point to do this with GN ?

Thanks

caipiblack, 2022-09-17:
This is just to make sure that the issue will not be closed. This feature is really required for the integration of matter library into external projects.

angelybo, 2022-12-12:
Hi, I am also looking for this - integration with external projects requires a lot of manual hand holding. 
I"ll open a PR for proof of concept when it's working.

woody-apple, 2023-06-23:
I don't believe this is being targeted any time soon. We already have many external projects using Matter as is. @angelybo or @caipiblack can you please elaborate what is blocking integration?

caipiblack, 2023-06-23:
Hello,

I don't know how "external projects" use the Matter stack and I will be curious to know.

--> Did they integrate their code **in the Matter build system** to build **their app** ?

Our idea is to be able to use the Matter library **like any library**. The only missing thing is "install" part that permits to install:

- headers in: {prefix}/include
- libraries in: {prefix}/lib
- pkgconfig files (.pc) in: {prefix}/lib/pkgconfig/

Basically most of the libraries do it, for exemple JsonCPP:

```
cd /tmp
git clone https://github.com/open-source-parsers/jsoncpp.git
cd jsoncpp/
mkdir build && cd build
cmake -DCMAKE_INSTALL_PREFIX:PATH=/tmp/install-dir ..
make
make install
```

Let's take a look at the install folder (`/tmp/install-dir`):

```
mathieu@mathieu-PC-Portable:/tmp/install-dir$ ls
include  lib
mathieu@mathieu-PC-Portable:/tmp/install-dir$ ls include/
json
mathieu@mathieu-PC-Portable:/tmp/install-dir$ ls include/json/
allocator.h   config.h    json_features.h  reader.h  version.h
assertions.h  forwards.h  json.h           value.h   writer.h
mathieu@mathieu-PC-Portable:/tmp/install-dir$ ls lib/
cmake         libjsoncpp.so        libjsoncpp.so.25  pkgconfig
libjsoncpp.a  libjsoncpp.so.1.9.5  objects-Release
mathieu@mathieu-PC-Portable:/tmp/install-dir$ ls lib/pkgconfig/
jsoncpp.pc
```

When we have something like this, then we are able to link a software A with a library B "easily" (without having to configure "include directories" that points to **another repository** and such things)

The `.pc` is not mandatory but usefull, it helps to specify include dirs, ldflags, cflags (...) required to build a software with the library.

We are integrating our project (and Matter) in OpenWRT and Yocto, in the two cases this is how it works:

- We have a "recipe" for our software that describe how to build it and what are the runtime or build dependencies. Matter is defined as a "build dependancy" (dependancy required to build our software)
- There is a "recipe" for each dependency that describe the same thing

When we build a software A with yocto or openwrt, the mechanism is able to know that the software require the libraries X,Y or Z, so first it build the libraries and then it install their headers/libs/pc in a folder and then it build the software A with correct cflags/ldflags to find/link headers/libs from this folder

When the "install" thing is included in the build mechanism (Cmake, GN..) of the project, it's perfect.

In our case as there is no "install" mechanism in the Matter library, we have to do it manually.

Attached ([matter-recipe.zip](https://github.com/project-chip/connectedhomeip/files/11847460/matter-recipe.zip)) you will find our recipe to build the "Matter" stack in Yocto it is based on the meta-matter from NXP but has been modified because in their example they just build the examples and their recipe cannot be used to provide the matter library and headers.

You will also find the tool that we made to do the "install" part "externally". Our tool is not really good because, to reduce maintenance we add includes from most of the directories, so we probably have useless includes exported.

---------------------------

To resume:

It should be good to implement something in the matter stack that permits to:
- Copy all required headers to link with libCHIP.a in {prefix}/include
- Copy all required libs in {prefix}/lib
- Generate a .pc file with required parameters in {prefix}/lib/pkgconfig

This thing should be integrated in the GN stuff idealy or on external script if it's not possible to do it with GN.

Doing it with GN instead of external script is better because: in the GN stuff, you are supposed to know the include files part of libCHIP.a, so you don't have to set manual folders like my script (avoid maintenance when folders are created/deleted/modified..)

NOTE: The "prefix" (path where to install `include` and `lib` folder) should be a parameter, like in the Cmake system "CMAKE_INSTALL_PREFIX"

---------------------------

ericriff, 2023-12-18:
I'd also like to see this, it would make this SDK (or at least the core lib) package-able.

gerickson, 2024-06-05:
+1 on this. I also mentioned this in my comment in #33521. Recapping that here and echoing @caipiblack's well-expressed comments above:

> I am just getting started with Matter myself and have noted some challenges with the build system as well. Questions you (@gaganpydi) might have encountered that I have encountered:
>
> 1. I cannot seem to find a way to specify: "use this C compiler, this C++ compiler, this linker, this archiver, these C flags, these C++ flags, and these linker flags". This is beyond trivial in autotools and straightforward enough in Meson or CMake.
> 2. I cannot seem to find a way to stage the build such that I get the binary artifacts and the public header files that define the public APIs for those binary artifacts. For example, under , I see many executables and many *.a's under lib/, but not headers to be found (this relates to @caipiblack's comments above).
> 3. I cannot seem to find a way to specify, "I already have the satisfied dependencies for nlassert, olio, openttread, ot-br-posix, etc. in my project. Build and link against them here, here, here, here, and here.". This is straightforward enough in CMake, autotools, or Meson.
> 4. I cannot seem to find a way to specify, "I don't want to use any of the platform definitions contained in the project. I will be supplying these externally. Either trust me and link against weakly-defined stubs or find them here." This is somewhat related to (2) in that if there were a stage-headers or install-headers target / goal, a platform / product integrator could run that first, build his/her platform against that, and then cycle back and build the rest of Matter against that. Building my platform / product integration in the Matter tree is a non-starter.

My expectation was that the core Matter library was going to build and function more like OpenSSL or some such or even like building an out-of-tree kernel module in Linux. So, far, that hasn't been my experience. However, I can also imagine this is new, uninitiated user error.

gerickson, 2024-06-05:
Per [this URL](https://gn.googlesource.com/gn/+/main/docs/reference.md#var_public), it looks like `gn` has a `public` declaration that, at minimum, should tag which headers are public.

I'll try instrumenting a few and see if that gets them propagated in any manner into the build artifacts tree in any way.

gerickson, 2024-06-05:
As an experiment, I made the following patch:

```diff
$ git diff .
diff --git a/src/messaging/BUILD.gn b/src/messaging/BUILD.gn
index a170d625111a..7a13425741e0 100644
--- a/src/messaging/BUILD.gn
+++ b/src/messaging/BUILD.gn
@@ -48,6 +48,10 @@ source_set("configurations") {
 static_library("messaging") {
   output_name = "libMessagingLayer"
 
+  public = [
+    "ExchangeContext.h"
+  ]
+
   sources = [
     "ApplicationExchangeDispatch.cpp",
     "ApplicationExchangeDispatch.h",
```

And then regenerated the project with:

```
$ gn --root=`pwd`/repo gen ${BuildRoot}/third_party/matter/.build/Makefile/${BuildProduct}/${BuildConfig} --args='is_debug=false'
```

and the built with:

```
$ ninja -j8 -C ${BuildRoot}/third_party/matter/.build/Makefile/${BuildProduct}/${BuildConfig}
```

but did not end up with _ExchangeContext.h_ in _${BuildRoot}/third_party/matter/.build/Makefile/${BuildProduct}/${BuildConfig}_.

Since they show up in the build artifacts, I thought that maybe _CHIPVersion.h_ or _ASN1OID.h_ might be good templates to follow; however, those are both generated by Python scripts with an explicit action.

So, I remain stumped at how to "stage" public headers for an interface or library.

gerickson, 2024-06-05:
I also tried this diff on the assumption that listing the header in `sources` was implicitly making it private and overriding the `public` declaration:

```diff
diff --git a/src/messaging/BUILD.gn b/src/messaging/BUILD.gn
index a170d625111a..feb72410c250 100644
--- a/src/messaging/BUILD.gn
+++ b/src/messaging/BUILD.gn
@@ -48,6 +48,10 @@ source_set("configurations") {
 static_library("messaging") {
   output_name = "libMessagingLayer"
 
+  public = [
+    "ExchangeContext.h"
+  ]
+
   sources = [
     "ApplicationExchangeDispatch.cpp",
     "ApplicationExchangeDispatch.h",
@@ -55,7 +59,6 @@ static_library("messaging") {
     "ErrorCategory.cpp",
     "ErrorCategory.h",
     "ExchangeContext.cpp",
-    "ExchangeContext.h",
     "ExchangeDelegate.h",
     "ExchangeHolder.h",
     "ExchangeMessageDispatch.cpp",
```

Unfortunately, that did not work either.

mspang, 2024-06-05:
Strongly recommend including headers _in situ_ from the main source tree since that's all that we're testing.

Although it's possible to distribute a library with a well specified interface and compiled code, the design of the Matter SDK isn't suited to that. Particularly, even the lowest level interfaces depend on build time configuration that is ABI affecting and that that may needs tuning not only on a platform basis but even on an application basis. Trying to install as if it were a reusable artifact would be very misleading.

gerickson, 2024-06-05:
Acknowledged that there are configuration "dials" for Matter (as there were for OpenWeave before it) that alter the ABI. I think that can be well-documented and -published such that there isn't any ambiguity about the stability of the ABI. That fact certain makes a strong argument against having a shared library option for the SDK.

That caveated, assuming that I wanted to add the public headers and interfaces (which in the source tree or out of the source tree are still the same) to the set of things that end up in the build artifacts folder, what is the right `gn` syntax to make that happen given that my experiments above with what seems to be the well-documented `public` approach didn't work?

In the interest of Matter growth and adoption, I think there exists an opportunity here to make Matter much more easily approachable and much more easily integrated into new products and platforms.

mspang, 2024-06-06:
You can list headers with a script like

gn gen out/host && gn desc out/host '*' --format=json >| matter.json
python3 headers.py

```
import functools
import json

gn_json = json.loads(open("matter.json").read())

def is_header(filename):
    return filename.endswith(".h") or filename.endswith(".hpp")

def public_headers(target):
    if target["public"] != "*":
        return target["public"]
    return filter(is_header, target.get("sources", []))

@functools.cache
def transitive_headers(target):
    target_json = gn_json[target]
    return set(public_headers(target_json)).union(
        *[transitive_headers(public_dep) for public_dep in target_json.get("deps", [])]
    )

for header in sorted(transitive_headers("//src/lib:lib")):
    print(header)
```

which produces

```
//out/host/gen/include/CHIPVersion.h
//out/host/gen/include/FirmwareBuildTime.h
//out/host/gen/include/app/AppBuildConfig.h
//out/host/gen/include/app/icd/server/ICDServerBuildConfig.h
//out/host/gen/include/asn1/ASN1OID.h
//out/host/gen/include/ble/BleBuildConfig.h
//out/host/gen/include/core/CHIPBuildConfig.h
//out/host/gen/include/crypto/CryptoBuildConfig.h
//out/host/gen/include/inet/InetBuildConfig.h
//out/host/gen/include/matter/tracing/build_config.h
//out/host/gen/include/platform/CHIPDeviceBuildConfig.h
//out/host/gen/include/platform/Linux/dbus/bluez/DbusBluez.h
//out/host/gen/include/platform/Linux/dbus/openthread/introspect.h
//out/host/gen/include/platform/Linux/dbus/wpa/DBusWpa.h
//out/host/gen/include/platform/Linux/dbus/wpa/DBusWpaBss.h
//out/host/gen/include/platform/Linux/dbus/wpa/DBusWpaInterface.h
//out/host/gen/include/platform/Linux/dbus/wpa/DBusWpaNetwork.h
//out/host/gen/include/setup_payload/CHIPAdditionalDataPayloadBuildConfig.h
//out/host/gen/include/system/SystemBuildConfig.h
//src/access/AccessControl.h
//src/access/AuthMode.h
//src/access/Privilege.h
//src/access/RequestPath.h
//src/access/SubjectDescriptor.h
//src/access/examples/ExampleAccessControlDelegate.h
//src/access/examples/PermissiveAccessControlDelegate.h
//src/app/AppConfig.h
//src/app/AttributeAccessInterface.h
//src/app/AttributeAccessInterfaceCache.h
//src/app/AttributeAccessInterfaceRegistry.h
//src/app/AttributeEncodeState.h
//src/app/AttributePathExpandIterator.h
//src/app/AttributePathParams.h
//src/app/AttributePersistenceProvider.h
//src/app/AttributeReportBuilder.h
//src/app/AttributeValueDecoder.h
//src/app/AttributeValueEncoder.h
[... 588 lines cut ...]
//third_party/nlio/repo/include/nlio-byteorder-big.hpp
//third_party/nlio/repo/include/nlio-byteorder-little.h
//third_party/nlio/repo/include/nlio-byteorder-little.hpp
//third_party/nlio/repo/include/nlio-byteorder.h
//third_party/nlio/repo/include/nlio-byteorder.hpp
//third_party/nlio/repo/include/nlio-private.h
//third_party/nlio/repo/include/nlio.h
//third_party/nlio/repo/include/nlio.hpp
//zzz_generated/app-common/app-common/zap-generated/attribute-type.h
//zzz_generated/app-common/app-common/zap-generated/cluster-enums-check.h
//zzz_generated/app-common/app-common/zap-generated/cluster-enums.h
//zzz_generated/app-common/app-common/zap-generated/cluster-objects.h
//zzz_generated/app-common/app-common/zap-generated/ids/Attributes.h
//zzz_generated/app-common/app-common/zap-generated/ids/Clusters.h
//zzz_generated/app-common/app-common/zap-generated/ids/Commands.h
//zzz_generated/app-common/app-common/zap-generated/ids/Events.h
```

This list will vary depending on how the build was configured. There will probably be something missing as listing headers in the build file is only weakly enforced in CI, but they can be added at the appropriate place.

Other settings relevant to library interface are also in the JSON such as defines and include paths and can be gathered a similar way.

The ZAP generated code is also going to require special care.

I still wouldn't encourage separate distribution of headers and sources. There isn't enough decoupling or layering or ABI attention to make precompilation worthwhile.

mspang, 2024-06-06:
In summary - making a tarball of what's compiled by GN isn't the hard part. The hard part is that to port Matter, you need to make significant changes to the SDK and recompile it.

tongjiang, 2024-12-12:
> You can list headers with a script like
>
> gn gen out/host && gn desc out/host '*' --format=json >| matter.json python3 headers.py
>
> ```
> import functools
> import json
>
> gn_json = [...]

Hi @caipiblack , for now, is there a "make install" command for us to get the export include path/library path/ldflags/cflags  of matter instead of dealing this by hand or external scripts.

caipiblack, 2024-12-12:
> Hi @caipiblack , for now, is there a "make install" command for us to get the export include path/library path/ldflags/cflags of matter instead of dealing this by hand or external scripts.

Hello,

As I know: No, I think the equivalent to "make install" does not really exists with GN.

I removed my message, because initialy I was thinking you tried to build the stack with yocto (on your issue), but after re-reading, it was not the case.

So the current build system is not perfect:
- Due to activate.sh that imply growing requirements (Ubuntu at specific version, python at specific version, etc)
- The GN stuff not providing install/staging.

These restriction are a real pain, when you need to build the stack in a building environment like Yocto/OpenWRT.

As suggested on the top, currently we do something like this:

```
source scripts/activate.sh

gn gen out/host --args='treat_warnings_as_errors=false chip_use_data_model_interface="enabled" chip_enable_wifi=false chip_enable_python_modules=false chip_enable_ble=false chip_config_network_layer_ble=false chip_enable_openthread=false config_enable_yaml_tests=false chip_build_tests=false chip_build_test_static_libraries=false'

ninja -C out/host

gn desc out/host '*' --format=json > /tmp/matter.json
```

And then we parse /tmp/matter.json with a python script to generate CMake Libs that we need.

The generating scripts looks something like this:

```

tongjiang, 2024-12-17:
@caipiblack Thanks to your python scripts, helps a lot.
But still have some intergration issue.
In my case, I need to build bridge-app, but I can't just using the Matter build system, cause I need to build this app using a lot outside thirdparty repositories.
I want to build the bridge app outside, So I just use the code in bridge-app and try to build it outside, the libs in the out/linux/lib is not enought, cause the app-main gn module is not build in out/linux/lib, so I create my own bridge-lib dir to build a libbridage.a to  out/linux/lib, then I can build my app outside the Matter project and using a lot outside thirdparty libs instead putting them all in Matter project. I suggest you guys can build the app-main project to a lib to let the developer just need to link this lib to developing there own apps outside the Matter project.

Thanks again!

tongjiang, 2024-12-18:
> Acknowledged that there are configuration "dials" for Matter (as there were for OpenWeave before it) that alter the ABI. I think that can be well-documented and -published such that there isn't any ambiguity about the stability of the ABI. That fact certainly makes a strong argument **against** having a shared library option for the SDK.
> 
> That caveated, assuming that I wanted to add the public headers and interfaces (which in the source tree or out of the source tree are still the same) to the set of things that end up in the build artifacts folder, what is the right `gn` syntax to make that happen given that my experiments above with what seems to be the well-documented `public` approach didn't work?
> 
> In the interest of Matter growth and adoption, I think there exists an opportunity here to make Matter much more easily approachable and much more easily integrated into new products and platforms. I'd be delighted to do the work and make the pull request(s); I just need some guidance on the `gn` magic to do so.

Hi @gerickson ,have you got any progress on this?
I have got the same problem as you, i want to intergrate matter to all project as a thirdparty component.
But Matter build system has not provide all the libs/include/cflags/ldflags for our project to intergrate.
I was developing the bridge app in our project, I thoughtthe libs in the out/targetxxx/lib directory is all the libs we need to intergrate, but I found that the libs is not enough. beside the app-main module was not compiled in library, there is still some issue.  I intent to make the bridge-app to a whole static library using "complete_static_lib = true", but still got some "undefined reference" function.

"/root/work/matter/out/linux/../../examples/bridge-app/linux/third_party/connectedhomeip/examples/common/tracing/decoder/interaction_model/DecoderCustomLog.cpp:74: undefined reference to `chip::Credentials::CsaCdKeysTrustStore::LookupVerifyingKey(chip::Span<unsigned char const> const&, chip::Crypto::P256PublicKey&) const'"