project-chip/connectedhomeip #1520: GN build parity - flashing

mspang, 2020-07-08:
GN build doesn't flash firmware.

Flashing is only useful for its side effects so it wouldn't be natural to flash from ninja, however an approach that can be used in situations like this is to generate a python script that will flash the firmware. Then the workflow becomes

  ninja -C out/firmware && out/firmware/flash_foo.py

kpschoedel, 2020-08-18:
I've been looking a bit at what Pigweed does and whether we could leverage something it already does. Sadly no, it's no less ad-hoc than the various Makefiles here, and there's no concept of flashing other than as an incidental step in the course of running a unit test. One interesting thing, though, is that for each unit test GN target it generates a ‘.testinfo.json’ with test metadata, which inspires me to think that we could have flashable GN targets generate a corresponding ‘.flash.json’ including the binary name and relevant toolchain- and target-specific data.

mspang, 2020-08-18:
I think having either a shell script or just keeping the current makefile and swapping out the build would unblock the migration (i.e., build with GN, make flash flashes the GN artifact). WDYT? If we want to improve it further we can do that afterwards.

kpschoedel, 2020-08-18:
Thinking transitionally, `make flash` would incompatible with the autotools `make flash`. Having the existing `make flash` transitionally invoke `gn` to build the flashable thing, instead of building makewise, would be possible but in general sounds like throwaway work to propagate make-based configuration values (for at least the esp32 case — the nrf5 is trivial).

closed 2020-09-08 by mspang