project-chip/connectedhomeip #1524: CHIPCryptoPALTest is slow

mspang, 2020-07-09:
Due to running 100 million iterations of PBKDF2. Do we need this many iterations to test the implementation?

grep iter src/crypto/tests/PBKDF2_SHA256_test_vectors.h
    unsigned int iter;
                                                                  .iter     = 1,
                                                                  .iter     = 2,
                                                                  .iter     = 4096,
                                                                  .iter     = 16777216,
    .iter     = 4096,
                                                                  .iter     = 16777216,
                                                                  .iter     = 16777216,
                                                                  .iter     = 16777216,
                                                                  .iter     = 16777216,
                                                                   .iter     = 16777216,
                                                                   .iter     = 16777216,

mspang, 2020-07-14:
@pan-apple

pan-apple, 2020-07-14:
@mspang , the last few test cases actually do not run that many iterations. The tests are expected to fail due to incorrect parameter. There's only one case where it runs 16777216 iterations.

mspang, 2020-07-14:
My mistake. Is it important for us to test the case of 16777216 iterations? If so maybe we could make that test optional. It's currently a significant factor in overall build times.

rwalker-apple, 2020-07-14:
I don't see how 16,777,216 iterations could be necessary.  We're just comparing a set of inputs to a golden output, right?  Could be 1 iteration?  Looks like our implementation is a simple pass-through to the underlying implementations which don't appear to special-case 16,777,216.

pan-apple, 2020-07-15:
These test vectors are commonly used for `pbkdf2` testing. Not sure if vector with 16,777,216 iterations is adding any value (or if it's worth the CI time). @balducci-apple do you have an opinion?

pan-apple, 2020-07-15:
> I don't see how 16,777,216 iterations could be necessary. We're just comparing a set of inputs to a golden output, right? Could be 1 iteration? Looks like our implementation is a simple pass-through to the underlying implementations which don't appear to special-case 16,777,216.

@rwalker-apple as you might have already noticed, the tests are written more as integration tests (instead of pure unit tests), where it's also testing the underlying library. I think it's worth testing it, since we have multiple crypto libraries (`OpenSSL`, `mbedTLS`), with slightly different/custom implementation of `mbedTLS` for different platforms.

rwalker-apple, 2020-07-15:
> These test vectors are commonly used for `pbkdf2` testing. Not sure if vector with 16,777,216 iterations is adding any value (or if it's worth the CI time).

Does that suggest that such a test has already been run on the imported implementations, or that clients of these libraries should run such a test?

pan-apple, 2020-07-15:
> Does that suggest that such a test has already been run on the imported implementations, or that clients of these libraries should run such a test?

These are just test vectors (given input => expected output). Anyone should be able to run these against a correct crypto implementation.

rwalker-apple, 2020-07-15:
> I don't see how 16,777,216 iterations could be necessary. We're just comparing a set of inputs to a golden output, right? Could be 1 iteration? Looks like our implementation is a simple pass-through to the underlying implementations which don't appear to special-case 16,777,216.

It occurs to me that in order to test our wrappers, we need to verify that there are no leaks.  The only place I can identify where a leak might occur is on repeated invocation (with error and successful exits) of the mbedtls version.  These cases aren't covered, AFAICT.

rwalker-apple, 2020-07-15:
> These are just test vectors (given input => expected output). Anyone should be able to run these against a correct crypto implementation.

ah, ok.  you mean the tests were copy-pasta'd?  that's cool, but perhaps we could generate a golden that's cheaper, since we should expect that the more expensive vectors have already been run on the imports.

pan-apple, 2020-07-15:
> ah, ok. you mean the tests were copy-pasta'd? that's cool, but perhaps we could generate a golden that's cheaper, since we should expect that the more expensive vectors have already been run on the [...]

Yeah, added @balducci-apple to the discussion.

pan-apple, 2020-07-15:
> It occurs to me that in order to test our wrappers, we need to verify that there are no leaks. The only place I can identify where a leak might occur is on repeated invocation (with error and [...]

Do you have a specific test condition? The test vectors do include some failure conditions.
I guess, you mean inducing errors by spoofing mbedtls/OpenSSL APIs.

rwalker-apple, 2020-07-15:
> Do you have a specific test condition? The test vectors do include some failure conditions.
> I guess, you mean inducing errors by spoofing mbedtls/OpenSSL APIs.

I'm looking at the ways that the CHIP mbedtls `pbkdf2_sha256()` implementation (which has provisions for mbedtls allocating on its behalf), might fail to clean up if the CHIP wrapper implementation ever changes.  Perhaps I've missed where CHIP checks that mbedtls doesn't leak if CHIP decides to abort an operation for invalid parameters.

balducci-apple, 2020-07-15:
> Yeah, added @balducci-apple to the discussion.

16777216 does indeed sound excessive :). Please assign this to me and I'll generate new vectors.

PR #1977 (2020-08-06): Reduce PBKDF2 iterations to speed up build (5x faster)
Reduce the number of PBKDF2 iterations from 16777216 to 3. This reduces
the MbedTLS build time from 44s to 8s on my machine.

fixes #1524
commit ebfe3b66 (2020-08-06): Make MbedTLS build 5x faster (#1977)

Reduce the number of PBKDF2 iterations from 16777216 to 3. This reduces
the MbedTLS build time from 44s to 8s on my machine.

fixes #1524
closed 2020-08-06 by BroderickCarlin