project-chip/connectedhomeip #1553: Build broken for ESP32 demo

mlepage-google, 2020-07-10:
#### Problem
Build broken for ESP32 demo:

/home/mlepage/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/mlepage/Projects/connectedhomeip/examples/wifi-echo/server/esp32/build/chip/lib//libDeviceLayer.a(libDeviceLayer_a-PlatformManagerImpl.o):(.literal._ZN4chip11DeviceLayer8Internal26GenericPlatformManagerImplINS0_19PlatformManagerImplEE16_AddEventHandlerEPFvPKNS0_15ChipDeviceEventEiEi[chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_AddEventHandler(void (*)(chip::DeviceLayer::ChipDeviceEvent const*, int), int)]+0x0): **undefined reference to `chip::Platform::MemoryAlloc(unsigned int)'**
/home/mlepage/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/mlepage/Projects/connectedhomeip/examples/wifi-echo/server/esp32/build/chip/lib//libDeviceLayer.a(libDeviceLayer_a-PlatformManagerImpl.o): in function `chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_AddEventHandler(void (*)(chip::DeviceLayer::ChipDeviceEvent const*, int), int)':
/home/mlepage/Projects/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.ipp:124: **undefined reference to `chip::Platform::MemoryAlloc(unsigned int)'**
collect2: error: ld returned 1 exit status
make: *** [/home/mlepage/local/esp-idf/make/project.mk:529: /home/mlepage/Projects/connectedhomeip/examples/wifi-echo/server/esp32/build/chip-wifi-echo.elf] Error 1

 #### Proposed Solution
Likely introduced by https://github.com/project-chip/connectedhomeip/pull/1427
Suspect new dependency was not added to ESP32 demo build

andy31415, 2020-07-10:
Got it working. In my case, I believe it is an autotools update issue:
  - we got new files in the build system
  - autotools does not regenerate .in files until bootstrap is called
  - by chance, code compiles except implementation for new stuff is not included in the build

to fix, re-run bootstrap, like in esp32 run `../../../../bootstrap`

Switching esp32 to gn would help a lot (although it seems ESP32 build system is reasonably complex, with all 'idf' integrations)

andy31415, 2020-07-10:
Could re-open for better documentation or wait for GN to have esp32 support. I wish for the latter.

closed 2020-07-10 by andy31415