project-chip/connectedhomeip #1558: NRF image fails to build

andy31415, 2020-07-10:
#### Problem

Image failed to build with:

```
+ wget -qO- https://www.nordicsemi.com/-/media/Software-and-other-downloads/Desktop-software/nRF-command-line-tools/sw/Versions-10-x-x/10-7-0/nRFCommandLineTools1070Linuxamd64tar.gz
+ tar xz -C /tmp
+ dpkg -i /tmp/JLink_Linux_V662b_x86_64.deb
Selecting previously unselected package jlink.
(Reading database ... 30635 files and directories currently installed.)
Preparing to unpack .../JLink_Linux_V662b_x86_64.deb ...
Removing /opt/SEGGER/JLink ...
/opt/SEGGER/JLink not found (OK)
Unpacking jlink (6.622) ...
dpkg: dependency problems prevent configuration of jlink:
 jlink depends on libncurses5 (>= 5.5); however:
  Package libncurses5 is not installed.

dpkg: error processing package jlink (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 jlink

```

 #### Proposed Solution
Add ncurses library to the image.

rwalker-apple, 2020-07-10:
ncurses is already part of the image, the problem is that debian doesn't honor ncurses6 as ncurses5 compatible.  the other nrf docker image has a fix for this.  wish these 2 images were combined.

andy31415, 2020-07-10:
I believe there is an open issue to merge. I will look into having a quick fix for now: android image seems to be stuck to 2.18 - need to make it 3.0 and nrf fails.

rwalker-apple, 2020-07-10:
yeah, this is a new Dockerfile from the Nordic folks that probably needed rebasing on my ubuntu:focal PR before landing

PR #1561 (2020-07-11): Update version for android image, fix dpkg install error in nrf conne…
…ct image

 #### Problem
NRf image fails to build. Android image is setting 0.2.18 version.

 #### Summary of Changes
- add force dpkg install for nrf build
- update android image to use a symlink for versioning

fixes #1558
closed 2020-07-11 by woody-apple