project-chip/connectedhomeip #1573: GN build parity: shell should build for linux (embedded) platform

turon, 2020-07-12:
#### Problem
Currently, a top-level build does not produce a `chip-shell` executable in:
`out/debug/linux_x64_gcc`
`out/debug/linux_x64_clang`
`out/debug/linux_x64_gcc_mbedtls`

 #### Proposed Solution
Add these targets to the platform=linux build.

mspang, 2020-07-13:
Is this a parity issue? The only build I see for shell is examples/shell/standalone/Makefile.

(Those other configs actually used to build these, but I removed them because it looked to me like it was only supposed to build with config/standalone/ ).

mspang, 2020-07-13:
This was the hunk:

diff --git a/BUILD.gn b/BUILD.gn
index a9ffe380..b68101c3 100644
--- a/BUILD.gn
+++ b/BUILD.gn
@@ -57,8 +57,6 @@ if (current_toolchain != "${dir_pw_toolchain}/dummy:dummy") {
 
     if (chip_build_tools) {
       deps += [
-        "${chip_root}/examples/chip-tool",
-        "${chip_root}/examples/shell",
         "${chip_root}/src/qrcodetool",
         "${chip_root}/src/setup_payload",
       ]

If you undo that, it will build in the remaining toolchains again.

turon, 2020-07-13:
@mspang Ok, I agree that calling this a "parity" issue may not be appropriate.

Code-wise, the shell example is valid for both `standalone` and `linux` targets.

For other embedded targets, we'll need some HAL UART abstraction which isn't ready yet.

PR #1574 (2020-07-14): [shell] Add device layer commands: start, get, config.
 #### Problem
linux-embedded build of shell example is missing from GN build.
Also, there are no commands yet to start the device layer or interact with the device configuration.

 #### Summary of Changes
Adds an initial linux-embedded target for GN.
Adds `device` command to shell, with subcommand `start`, `config`, and `get`.

fixes #1573, fixes #1586
closed 2020-07-14 by woody-apple