project-chip/connectedhomeip #1590: I need echo profile to develop/debug networking and ChipOverble work

yunhanw-google, 2020-07-14:
#### Problem
I am working on integrating BlueZ BLE in CHIP device layer, and needs to test echo message upon message layer between mock-client and mock-server, and network latency. 

In CHIP, there is no dedicated general echo profile. I think echo profile would be beneficial for lots of network test and debugging between mock client and server in complicated IOT topology.

 #### Proposed Solution
I would like to port Weave ECHO profile to CHIP, which is pretty lightweight and provides a basic mechanism to test network connectivity and latency.  It is modeled after the ICMP echo request and is intended for similar uses.
https://github.com/openweave/openweave-core/tree/master/src/lib/profiles/echo/Current

kghost, 2020-07-14:
The purpose is not implement the echo profile. the echo profile just act as example or demo for people to write other profiles.

The major work is not the echo profile. We need a dispatcher to dispatch messages from transports or secure channels to profiles, and provides an API to be used by profiles to send messages.

woody-apple, 2020-07-22:
@yunhanw-google We already have an echo like server, and soon to be on/off clusters. What is this covering?

yunhanw-google, 2020-07-22:
@woody-apple  I think that is examples/wifi-echo/esp32/main/EchoServer.cpp, I would like to have the dedicated echo profile that can be used by echo client and echo server in mock-device and linux device layer, esp32, darwin. Current EchoServer.cpp does not reply the real echo message.

For example, in existing weave mock-test framework, we bring up mock-client and server, and send echo message with and without security during presubmit, and also we extend it to different platform, esp32, linux, and darwin.

Since weave echo profile is pretty light-weight, https://github.com/openweave/openweave-core/tree/master/src/lib/profiles/echo/Current,  after we port it, we would update esp32/main/EchoServer.cpp

For ZCL data model part, I think every cluster needs to use the common dedicated echo profile to chat. Also Echo should also work without cluster, it should be generic, separate one, for example, if I just wanna evaluate thread, chip over ble and wifi or security stuff, I can use dedicate echo client and server to do that.

yunhanw-google, 2020-07-25:
After offline discussion, we agree to port weave echo profile to ZCL cluster in CHIP.
Since echo profile is the first and simplest weave profile we start to port, we expect this would act as one good reference port for the other profiles.

Please help assign this ticket to @kghost who is working on this.

Thank you.

closed 2021-09-29 by stale[bot]