project-chip/connectedhomeip #1595: Standardize how build features are included

andy31415, 2020-07-14:
#### Problem
Existing pattern for including features is to
  - have a #define in a main config file
  - use #ifdef to comment/uncomment an entire .cpp file from the build
  - build system includes all files to compile, relying on that only some files will actually include the functionality based on #ifdefs

This seems confusing:
  - Including 'foo.cpp' in the build does not guarantee that foo is available
  - compiles more files than necessary 
  - some compilers issue warnings if the a compilation output is actually empty

 #### Proposed Solution
  - Let the build system include necessary files. We can provide #ifdefs as well, however users of a library should not need to know how an interface is provided as long as something is available (e.g. memory interface)

turon, 2020-07-14:
As stated in #1592 

Large systems such as CHIP tend to have many ways to pass configuration:

1) through the build system (thus building hard dependencies and maintenance requirements on those build systems that are supported)
2) through C defines and config.h headers
3) both

In order to better decouple the codebase from any particular build system, OpenThread as a project had agreed upon and moved to a model where most configuration was passed via (2).  What is nice about (2) is it simplifies the build system and thus better supports easier scaling to multiple build systems.  It is also easier to build the portions of the CHIP project with an application project-specific build system (ideally converging to just including all necessary files and include paths, and configure the package as needed via a `CHIPProjectConfig.h`).

gerickson, 2020-07-20:
The downside, of course, of moving configuration to C/C++ header files, unlike `Kconfig` (Linux, ESP32, U-Boot) or `configure` is that you tend to lose discoverability and auto-configuration. Trade-offs, as always.

sweetymhaiske, 2021-07-06:
@andy31415  Does this issue needs to keep open?

closed 2022-09-22 by stale[bot]