project-chip/connectedhomeip #1633: Decide whether the size of a PacketBuffer is size_t or uint16_t

bzbarsky-apple, 2020-07-16:
#### Problem
We have the following function on `PacketBuffer`:
```
    uint16_t DataLength(void) const;
    void SetDataLength(uint16_t aNewLen, PacketBuffer * aChainHead = NULL);
    uint16_t AvailableDataLength(void) const;
    static PacketBuffer * NewWithAvailableSize(size_t aAvailableSize);
    static PacketBuffer * NewWithAvailableSize(uint16_t aReservedSize, size_t aAvailableSize);
```
so it generally seems like data length is `uint16_t` but available size bits are `size_t`, except `AvailableDataLength` also returns `uint16_t`.  Which is it?  Is `PacketBuffer` expected to support sizes that don't fit into `uint16_t`?

 #### Proposed Solution
Decide what sizes we actually support, change types as needed, and maybe document clearly.

@pan-apple @rwalker-apple @bhaskar-apple @gerickson

gerickson, 2020-07-21:
Historically, in the context of Weave, we have never had an individual packet or message larger than 64 KiB, so `uint16_t` has sufficed. This also aligns well with the maximum size of a UDP datagram. Finally, LwIP has limited its packet buffers to 16-bits of length. So, from a portability perspective, we'd have to address that port limitation as well.

The best approach might be to leave things at uint16_t, but to have a public PacketBuffer typedef of `Length` or some such that then use that typedef'd size everywhere rather than a bare `uint16_t`.

If we envision that something larger is needed / warranted, then we can look at a different type.

gerickson, 2020-07-21:
In short, I'd recommend `uint16_t`.

bzbarsky-apple, 2020-09-03:
Related: should `MessageHeader` use `uint16_t` for things like `Decode` and its sizing in general?

I tried enabling `-Werror=conversion` for https://github.com/project-chip/connectedhomeip/issues/1641 and immediately ran into `size_t` values being passed to functions declared as taking `uint16_t`.

rwalker-apple, 2020-09-03:
I don't think we get a choice here, unless we de-couple PacketBuffer from lwip, which uses uint16_t

bzbarsky-apple, 2020-09-30:
> Related: should MessageHeader use uint16_t for things like Decode and its sizing in general?

I decided the answer to this is "yes", after completely forgetting about it for a bit...  https://github.com/project-chip/connectedhomeip/pull/2836 fixed this.

closed 2020-10-01 by bzbarsky-apple