project-chip/connectedhomeip #1636: Request to check Build Android

jonggab-park, 2020-07-17:
We try to build Android with your building guide, but we meet the build error.

This is my environment and version of tools.
export JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64
export ANDROID_HOME=~/Android/Sdk
export ANDROID_NDK_HOME=~/Android/Sdk/ndk/21.3.6528147

After downloading codes on master branch, I just run 'make -f Makefile-Android'

Could you check to build Android??

woody-apple, 2020-09-21:
@vidhis88 ?

vidhis88, 2020-09-22:
The steps to build Android are listed in the README here: https://github.com/project-chip/connectedhomeip/tree/master/src/android/CHIPTool
Please let me know if this still doesn't help build the Android demo app.

andy31415, 2020-10-20:
I believe this issue is obsolete: we moved the build system to GN and there are build instructions.

The error seems to be caused by a failure to load so (either themselves or libstdc++). This should be resolved if following the readme since it contains some not as obvious steps.

closed 2020-10-20 by andy31415