project-chip/connectedhomeip #1652: Add the CHIP Profile ID to the transport message header

shana-apple, 2020-07-20:
We will be introducing something similar to weave profiles so adding the profile id to the message header.

Note from @pan-apple :  The profile ID will eventually be in the encrypted section of the message header. Right now it is being added in the non-encrypted part. That’s ok for time being, as we still need to finalize the message header format and encrypt relevant portions.

Follow up questions:
Should this be called a profile ID/ CHIP Profiles? "Profiles" are also a ZCL concept so should we change it to something different to avoid confusion.

gerickson, 2020-07-20:
/cc/ @yufengwangca

yunhanw-google, 2020-07-22:
Hi, 
We are porting weave exchange manager to CHIP, which route packet to different profile including echo, network provisioning, zcl.
@yufengwangca @gerickson @kghost 
https://github.com/project-chip/connectedhomeip/issues/1670

yufengwangca, 2020-07-23:
This issue seems has been covered by https://github.com/project-chip/connectedhomeip/issues/1692

woody-apple, 2020-07-23:
Agree, duplicate of #1692

closed 2020-07-23 by woody-apple