project-chip/connectedhomeip #1668: Unable to build the  source tree

manuel-apple, 2020-07-22:
I  am following  the instructions to setup  the development environment using  Visual Code.
All  initial steps  to  setup development container  and run bootstrap work.

When I  get to the last step,  I  do Run Build Task ---> Main. Terminal shows it is compiling but then  stops with  an  error:

checking D-Bus configuration directory... /etc
checking D-Bus system bus services dir... /usr/share/dbus-1/system-services
checking D-Bus session bus services dir... /usr/share/dbus-1/services
${prefix}/share/zsh/site-functions
checking for UDEV... no
configure: error: libudev >= 172 is required
configure: error: ../../../../../third_party/bluez/repo/configure failed for third_party/bluez/repo
The terminal process "/bin/bash '-c', 'scripts/build/bootstrap.sh'" terminated with exit code: 1.

I  am using a Mac for development

turon, 2020-07-22:
@yufengwangca could this be related to #1557?

yunhanw-google, 2020-07-22:
You should not compile linux device layer in MacOS

configure --with-device-layer=none

yunhanw-google, 2020-07-22:
May I see your mac compilation instruction? thanks

manuel-apple, 2020-07-22:
I  followed the instructions from this page: https://github.com/project-chip/connectedhomeip/blob/master/docs/VSCODE_DEVELOPMENT.md

pan-apple, 2020-07-22:
To add to it, the VSCode devcontainer is not running native MacOS build. It's an Ubuntu container. If I change the VSCode `Bootstrap` task with `--with-bluez=no`, the build works fine. I am not certain if this is a solution.

pan-apple, 2020-07-22:
I have pushed a draft PR. I can abandon the PR if there's a more comprehensive solution to the issue.

woody-apple, 2020-07-22:
Resolved in this PR, sorry @manuel-apple !

closed 2020-07-22 by woody-apple