project-chip/connectedhomeip #1669: Reland weave message layer from weave message layer to CHIP

yunhanw-google, 2020-07-22:
Weave message layer is broadly agreed in previous TSG meeting, we are trying to reland weave message layer to CHIP in order to prepare the integration for weave event logging, Weave BTP and so on.

yunhanw-google, 2020-07-23:
This is high-level ticket. We are still investigating the WML features needed for CHIP to align with message layer baseline, and try to incrementally create the meaningful feature ticket later.
currently we have the below 
https://github.com/project-chip/connectedhomeip/issues/1692
https://github.com/project-chip/connectedhomeip/issues/1670
thanks

woody-apple, 2020-07-23:
Let's keep issues actionable, as we start to put dates on things. closing this, given the other two.

closed 2020-07-23 by woody-apple