project-chip/connectedhomeip #1670: Port weave exchange manager and exchange context to CHIP

yunhanw-google, 2020-07-22:
In order to make weave profiles, for example, network provisioning, weave event logging, echo profile, we need to port weave exchange manager and exchange context

woody-apple, 2020-07-23:
I'm confused as to this. We should make clusters for these things, these live above the Secure Transport API, no need to handle transport here. Per our discussion re: Echo Profile -> Echo Cluster, same for other Clusters we need to update or create.

gerickson, 2020-07-27:
There should exist in CHIP both clusters (publish and subscribe semantics) and the ability to use the message layer to compose arbitrary protocols that are not clusters.

andy31415, 2020-08-05:
I believe 'exchange manager' == 'session manager' in the current CHIP source repo. We need to add missing semantics to it.

@yunhanw-google Could you add a list of missing functionality or what exchange manager purpuse is?

woody-apple, 2020-10-21:
@yunhanw-google Given you're working on this now, can you carve this issue into a couple things to reflect the current architecture? Thanks!

closed 2020-10-21 by woody-apple