project-chip/connectedhomeip #1672: Add mock device framework in CHIP

yunhanw-google, 2020-07-22:
In weave, we have mock-device which can achieve light-weight weave feature validation between mock client and mock server, for example, echo profile, network provisioning, and event validation.
We can take that mock-device framework from weave to CHIP and improve the code quality.

woody-apple, 2021-01-25:
SW BRB: What is the proposal here, and can this be cut up into more specific work? @erjiaqing ?

erjiaqing, 2021-01-26:
> SW BRB: What is the proposal here, and can this be cut up into more specific work? @erjiaqing ?

I think this can be replaced by happy framework / docker container and linux example app, since the linux example app is widely used for integration testing. Maybe this issue can be closed now.