project-chip/connectedhomeip #1690: Can't compile nrf5 lock app since submodule update of openthread without manually mucking with the submodule source tree

bzbarsky-apple, 2020-07-22:
#### Problem
When I try to follow the instructions in `examples/lock-app/nrf5/README.md`, whether using docker or not, I get a build error like so:
```
make[7]: *** No rule to make target 'backbone_router/leader.cpp', needed by 'backbone_router/libopenthread_ftd_a-leader.o'.  Stop.
```
I poked around a bit, and the openthread commit at https://github.com/openthread/openthread/commit/daf89169c66387dbbc38106abbac67e8904e27e2 renamed that file.  It changed the `src/core/Makefile.am` file in the openthread repo, but looking at our tree, the `src/core/Makefile.in` file still had the old filename for me.  We pulled that commit into our tree in https://github.com/project-chip/connectedhomeip/pull/1583 as far as I can tell.

Rerunning `scripts/build/boostrap.sh` does _not_ regenerate the `Makefile.in` in question.

If I manually delete `third_party/openthread/repo/src/core/Makefile.in` and then do the nrf5 lock-app build, things work fine...

 #### Proposed Solution
I'm not sure.  Presumably something here (either our bootstrap or the nrf5 build) should regenerate the relevant `Makefile.in`.

@mspang @rwalker-apple

mspang, 2020-07-24:
This is a known flaw in the automake builds.

I *think* @andy31415 found that configuring  --enable-maintainer-mode helped with this kind of problem.

mspang, 2020-09-01:
This is obsolete with GN

closed 2020-09-01 by mspang