project-chip/connectedhomeip #1692: [transport] Align MessageHeader format to Weave Message Layer baseline

turon, 2020-07-22:
#### Problem
Given the[ Weave Message Format specification](https://github.com/openweave/openweave-core/blob/master/doc/specs/weave-message-format.pdf) as baseline, the CHIP [Message Header](https://github.com/project-chip/connectedhomeip/blob/master/src/transport/MessageHeader.cpp#L30) has reordered some fields and is missing the Exchange fields (such as Profile ID, etc.)

 #### Proposed Solution
Recast the CHIP Message Header fields to align with the baseline specification.  Also update all tests and examples as appropriate to behave as before but with the updated line format.

```
[BEFORE]
```
/**********************************************
 * Header format (little endian):
 *
 *  16 bit: | VERSION: 4 bit | FLAGS: 4 bit | RESERVED: 8 bit |
 *  16 bit: | Secure message type                             |
 *  32 bit: | MESSAGE_ID                                      |
 *  32 bit: | Secure Session ID                               |
 *  64 bit: | Message Authentication Tag                      |
 *  64 bit: | SOURCE_NODE_ID (iff source node flag is set)    |
 *  64 bit: | DEST_NODE_ID (iff destination node flag is set) |
 *
 **********************************************/
 ```
 
[AFTER -- Weave compatible]
```
/**********************************************
 * Header format (little endian):
 *
 *  16 bit: | VERSION: 4 bit | FLAGS: 4 bit | RESERVED: 8 bit |
 *  32 bit: | MESSAGE_ID                                      |
 *  64 bit: | SOURCE_NODE_ID (iff source node flag is set)    |
 *  64 bit: | DEST_NODE_ID (iff destination node flag is set) |
 *   8 bit: | Exchange Header                                 | +add
 *   8 bit: | Exchange Message Type                           | (formally 16-bit Secure message type)
 *  16 bit: | Exchange ID                                     | (formally 32-bit Secure Session ID)
 *  32 bit: | Message Profile ID                              | +add
 *  64 bit: | Message Authentication Tag                      |
 **********************************************/
 ```

woody-apple, 2020-07-22:
Assigning to @pan-apple , given he's working on this.

woody-apple, 2020-07-22:
Note @pan-apple Not sure what the ProfileID is for yet, given clusters and that information are contained inside the secure payload. We should confirm with @balducci-apple and @RyanTheOptimist what the expectation is here

pan-apple, 2020-07-22:
@woody-apple I believe 32 bit profile ID is a mix of 16 bit vendor ID and 16 bit profile ID. The 16 bit profile ID is further used to distinguish between various logical blocks which need to handle the incoming messages (e.g. Rendezvous, OTA updates, ZCL cluster messages etc).

woody-apple, 2020-07-22:
Understood, those should be encapsulated inside the encrypted body. Not transparent on the wire :)

pan-apple, 2020-07-22:
Yes, this part of the header is expected to be encrypted. 
```
 *   8 bit: | Exchange Header                                 | +add
 *   8 bit: | Exchange Message Type                           | (formally 16-bit Secure message type)
 *  16 bit: | Exchange ID                                     | (formally 32-bit Secure Session ID)
 *  32 bit: | Message Profile ID                              | +add
 *  64 bit: | Message Authentication Tag                      |
```

woody-apple, 2020-07-22:
Updating Layer -> Format

turon, 2020-07-22:
I didn't include in this issue the security fields, but they would presumably be inserted before the Exchange/Session fields as per the WML spec.

pan-apple, 2020-07-22:
@turon what field identifies the key that was used for encrypting the message? WML seems to have a `key ID`. In the current header, we were thinking of using `Secure Session ID`. If Session fields are encrypted, then we cannot use it to identify the encryption keys.

turon, 2020-07-22:
[Future -- add security fields]
```
/**********************************************
 * Header format (little endian):
 *
 *  16 bit: | VERSION: 4 bit | FLAGS: 4 bit | RESERVED: 8 bit |
 *  32 bit: | MESSAGE_ID                                      |
 *  64 bit: | SOURCE_NODE_ID (iff source node flag is set)    |
 *  64 bit: | DEST_NODE_ID (iff destination node flag is set) |
 *. 16 bit: | Key ID.                                         | < +security header fields
 *. 16 bit: | Payload Length                                  | < +security header fields
 *. <var>:  | IV                                              | < +security header fields
 *   8 bit: | Exchange Header                                 |
 *   8 bit: | Exchange Message Type                           | (formally 16-bit Secure message type)
 *  16 bit: | Exchange ID                                     | (formally 32-bit Secure Session ID)
 *  32 bit: | Message Profile ID                              |
 *  64 bit: | Message Authentication Tag                      |
 **********************************************/
```

yufengwangca, 2020-07-23:
Yes, we can not used Secure Session ID (which should be renamed to Exchange ID) to identify the encryption keys. Weave use Key ID for this purpose.

Logically, the fields bellow IV (Initialization Vector) should be refactored out of MessageHeader, since this part is only used to identify the profile in which the message type is defined. Splitting this part to another header can allow us define CHIP message not targeted to a profile. Such as message sent to the cloud.

pan-apple, 2020-07-23:
Thanks @yufengwangca. Can you also confirm what fields in the proposed message header is encrypted (if any)?

yufengwangca, 2020-07-23:
The encryption should start from Exchange Header.

By the way, how the header format is defined, not sure if 4 bits are enough for all the different Flags .

16 bit: | VERSION: 4 bit | FLAGS: 4 bit | RESERVED: 8 bit

pan-apple, 2020-07-23:
> The encryption should start from Exchange Header.
> 
> By the way, how the header format is defined, not sure if 4 bits are enough for all the different Flags .
> 
> 16 bit: | VERSION: 4 bit | FLAGS: 4 bit | RESERVED: 8 bit

Thanks @yufengwangca, that aligns with my understanding. So the following fields are encrypted (and, the `Message Authentication Tag` is NOT encrypted, as it's needed for the decryption)

```
 *   8 bit: | Exchange Header                                 | +add
 *   8 bit: | Exchange Message Type                           | (formally 16-bit Secure message type)
 *  16 bit: | Exchange ID                                     | (formally 32-bit Secure Session ID)
 *  32 bit: | Message Profile ID                              | +add
```

Where does the application data (encrypted) wrt to the header? Typically it precedes the header. If we were to do that In this case, the encrypted blocks will be split into two parts. 

Header fields that are not encrypted
```
 *  16 bit: | VERSION: 4 bit | FLAGS: 4 bit | RESERVED: 8 bit |
 *  32 bit: | MESSAGE_ID                                      |
 *  64 bit: | SOURCE_NODE_ID (iff source node flag is set)    |
 *  64 bit: | DEST_NODE_ID (iff destination node flag is set) |
 *. 16 bit: | Key ID.                                         | < +security header fields
 *. 16 bit: | Payload Length                                  | < +security header fields
 *. <var>:  | IV                                              | < +security header fields
```

Header fields that are encrypted
```
 *   8 bit: | Exchange Header                                 |
 *   8 bit: | Exchange Message Type                           | (formally 16-bit Secure message type)
 *  16 bit: | Exchange ID                                     | (formally 32-bit Secure Session ID)
 *  32 bit: | Message Profile ID                              |
```

Message auth tag which is not encrypted
```
 *  64 bit: | Message Authentication Tag                      |
```

```
Encrypted application data
```

We can easily solve it by moving `Message Authentication Tag` before `Exchange Header`. Thoughts?

yufengwangca, 2020-07-23:
@pan-apple Could you educate me what 'Message Authentication Tag' is used for?  is this used for integrity check?

pan-apple, 2020-07-23:
@yufengwangca , the Tag is generated at data encryption (`Encrypt()` outputs the Tag). The tag is provided to `Decrypt()` as input parameter when the data is being decrypted. You are right, it's used for integrity check.

yufengwangca, 2020-07-23:
Thanks for conform, if 'Message Authentication Tag' is the integrity check value, then we need to store it immediately after 'Encrypted application data', following the end of all encrypted payload, do you agree?

yufengwangca, 2020-07-23:
Yes, there is not specific reason to put checksum after the encrypted payload except match Weave Message Format baseline.

yufengwangca, 2020-07-23:
@pan-apple Just checked the current implementation of Encrypt function. It forces the tag to be a type of unit64_t.

CHIP_ERROR SecureSession::Encrypt(const unsigned char * input, size_t input_length, unsigned char * output, MessageHeader & header)
{
 ......
    uint64_t tag     = 0;

    error = AES_CCM_encrypt(input, input_length, NULL, 0, mKey, sizeof(mKey), (const unsigned char *) &IV, sizeof(IV), output,
                            (unsigned char *) &tag, sizeof(tag));
    header.SetTag(tag);
   .......
}

But the output of tag from AES_CCM_encrypt has variable length. In Weave, the Message Integrity Check is stored immediately after the encrypted payload since the length of it is variable, and depends on the integrity check algorithm  in use. 

Do you know if it is intentional  to have the fixed length of Auth tag?

pan-apple, 2020-07-24:
Even in CHIP the length of tag can be 8, 12, or 16 bytes. So the length of tag will change from `uint64_t`. It's ok to put it after encrypted data. The structure definition in the issue did not show the positioning of encrypted data. So, just wanted to clarify.

yufengwangca, 2020-07-24:
Then we also need to introduce 'Encrypt Type' in the Header so that we can refer the size of IV, and  'Message Authentication Tag' fields.

The 'Payload Length' field is optional, it is only present when the selected encryption algorithm requires the use of message padding, current CHIP crypto function does not requires the use of padding. Thus is field is defined for future use only.

The  'IV' field is also optional, it is only present then the selected the selected encryption algorithm requires the use of 'Initialization Vector'.

So the final version should looks like this:

 Header fields that are not encrypted

 *  16 bit: | VERSION: 4 bit | FLAGS: 4 bit | Encrypt Type: 4 bit | RESERVED: 4 bit 
 *  32 bit: | MESSAGE_ID 
 *  64 bit: | SOURCE_NODE_ID (if source node flag is set)   
 *  64 bit: | DEST_NODE_ID (if destination node flag is set) 
 *  16 bit: | Key ID
 *  16 bit: | Payload Length
 *  Variable: | IV

Header fields that are encrypted

 *   8 bit: | Exchange Header 
 *   8 bit: | Exchange Message Type                                           // (formally 16-bit Secure message type)
 *  16 bit: | Exchange ID                                                               // (formally 32-bit Secure Session ID)
 *  32 bit: | Message Profile ID 
 *  Variable:  | Application data

Integrity checksum are not encrypted
 *  Variable:  | Message Authentication Tag

pan-apple, 2020-07-24:
The current code has `SessionID` in it, which is supposed to identify the encryption type.

yufengwangca, 2020-07-24:
Both Session ID and Key ID have their own dedicated purposes, they should not be reused to identify the encryption type.

Session ID (To be renamed to Exchange ID) is used to identify the unique conversation to which the message belongs, Key ID is used to identify the encryption integrity keys used encrypt the message. 

'Encrypt Type' is needed since it decides the availability of some fields and the length of certain fields in the CHIP message.

pan-apple, 2020-07-24:
> Both Session ID and Key ID have their own dedicated purposes, they should not be reused to identify the encryption type.
>
> Session ID (To be renamed to Exchange ID) is used to identify the unique [...]

I was using this comment as the reference: https://github.com/project-chip/connectedhomeip/pull/1519#discussion_r457663453

turon, 2020-07-30:
@pan-apple That comment seems to be about using `SessionID` to determine the `Tag` length.  That seems okay to me.  Combining `SessionID` and `KeyID` isn't proposed in that comment however, and keeping those two fields separate is preferred.

PR #1950 (2020-08-04): Update message header structure
 #### Problem
The message header structure wire format is missing some fields. And, some fields are at incorrect offsets.

 #### Summary of Changes
Update the message header structure to follow agreed upon format. Also, a part of structure is expected to be encrypted. Added support for the encrypted header fields.

fixes #1692
closed 2020-08-04 by woody-apple