project-chip/connectedhomeip #1713: linux_embedded and main" build are the same

mspang, 2020-07-23:
Now that linux device layer is enabled by default, the "main" and "linux_embedded" builds tested by CI are now identical.

To test the standalone build we must pass --with-device-layer=none (autotools) or device_platform="none" (GN).

mspang, 2020-07-23:
Reconfiguring these builds raises an issue - "main" suggests it is the default configuration, but that's no longer the case.

woody-apple, 2020-07-23:
@turon I thought linux_embedded was meant to be the accessory build, vs our main build is the controller side?

turon, 2020-07-23:
> @turon I thought linux_embedded was meant to be the accessory build, vs our main build is the controller side?

High-level, that assumption is valid in the current model.

The main distinction is that Device Layer assumes a device which auto-manages its interfaces and handles provisioning of those interfaces (i.e. WiFi, Thread, and BLE) through a CHIP product onboarding flow.

The "standalone" or "--with-device-layer=none" is a more low level construct which assumes pre-configured IP interfaces such as Thread or WiFi are already setup, assumed to be available, and can just be used.  Standalone assumes a POSIX platform and the application will directly provide and manage its own `select` loop.

Simplifying to one model would be great, but will require some design work to make the device layer more configurable so, for example, some modules such as `WiFiConnectivityManager` can be disabled for a controller build that doesn't need it or manages WiFi connectivity outside of a CHIP provisioning context.

mspang, 2020-07-24:
Martin, which should be the default in your opinion?

One question about the architecture I'm confused about - How is BLE ported for a controller if not via the BLEManager interface in src/platform ?