project-chip/connectedhomeip #1723: Support ACL setup

franck-apple, 2020-07-23:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

mlepage-google, 2022-01-27:
Hi, on "accessory side" you mean the client device (e.g. speaker hub or phone), which must be an administrator (to change ACLs), desiring to change ACLs on the server device (e.g. the light bulb)?

If so, that will occur through regular read/write attribute IM calls, and already functions now (e.g. using chip-repl) modulo a few details (like the fabric-filtered writes which are in the process of landing in the IM) and the fact that enforcement isn't turned on yet (can't do that until you can write ACLs and administrators are updated to do so).

I think this is a redundant issue and can be closed.

woody-apple, 2022-02-03:
Agree this can be closed.

closed 2022-02-03 by woody-apple