project-chip/connectedhomeip #1803: [Secure Channel] Timed Writes

franck-apple, 2020-07-24:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

pan-apple, 2020-08-31:
Timed write commands will be tagged with an expiry TTL. If an accessory mandates a timed write for an attribute, and it receives a timed write command, it tests that the TTL is not expired before executing the command. If the TTL is expired, it does not execute the command, and possibly returns an error. Also, if it receives a non-timed write command for the attribute, it fails the command, and returns an error.

woody-apple, 2021-01-25:
SW BRB: Assigning to @bzbarsky-apple

bzbarsky-apple, 2021-12-06:
Fixed in https://github.com/project-chip/connectedhomeip/pull/12567

closed 2021-12-06 by bzbarsky-apple