project-chip/connectedhomeip #1808: [Secure Channel] Device Node ID Map to IPv6 address - Implement API

franck-apple, 2020-07-24:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

Damian-Nordic, 2021-04-13:
Note that we have already implemented the `Resolver` interface based on Linux (avahi), Darwin. Also, the minimal mDNS  implementation is pending (PR created by Andy). Hence, I think the task may be too general and needs clarification.

gjc13, 2021-04-14:
The resolver interface shall suffice for implementing the node ID mapping mechanism. 
If this feature is required on end devices, Ww need to implement the `Resolver` interface. Then https://github.com/project-chip/connectedhomeip/pull/5907 shall be merged. We also need to add node ID resolution implementation for other platforms and OpenThread SRP.
As far as I'm concerned, currently all the ZCL subscritions originates from the controller. Is node ID mapping a must for end devices? If not, we can leave the interface unimplemented. This will reduce development effort especially given the fact that resolving services on WiFi is not yet imeplmented in OpenThread SRP.
@erjiaqing who is working on the interaction model may shed some light on this topic.

Damian-Nordic, 2021-04-15:
Ok. So I think it's quite safe to say that on the controller devices we've got a common mechanism to resolve and update device addresses (we still need to combine it with CRMP so that a DNS-SD query is generated on the first retransmission), but what we're missing is a common interface for accessories. Should we change the title of this issue to something like "Resolve Node ID to IP address on accessory devices"?

franck-apple, 2021-04-15:
Moving to TE3 milestones, as Accessory to Controller resolving is only needed for TE3

closed 2022-02-07 by woody-apple