project-chip/connectedhomeip #1829: [Provisioning Devices] Access Control - ACL provisioning

franck-apple, 2020-07-24:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

smides-nest, 2020-11-01:
This should have a "provisioning" label instead of "rendezvous" label. For some reason I can't edit.

mlepage-google, 2022-01-27:
Hi what is an accessory? What does this issue mean? Is this about administrators making the ACLs on behalf of the devices on their fabric, so they can operate on their fabric? If so, this issue will have to be assigned to whoever is responsible for those administrators.

andy31415, 2022-01-31:
I believe this was a catch-all bug per features, however specifically for ACL we have separate tracking at the moment and this bug is not needed. Closing.

closed 2022-01-31 by andy31415