project-chip/connectedhomeip #1936: linux: Add instructions for the process for generating these sources from the xml be added to the Linux/README.md

woody-apple, 2020-07-31:
Ack, will do after we finalize the WiFi support

_Originally posted by @yufengwangca in https://github.com/project-chip/connectedhomeip/pull/1909#discussion_r463410990_

mspang, 2020-08-30:
GN build generates these files on the fly. Once automake is removed they can be deleted.

yufengwangca, 2020-08-30:
@mspang 
What files need to be deleted since those files are generated during the build in GN?

    Linux/gen/DBusWpa.c                   \
    Linux/gen/DBusWpaInterface.c          \
    Linux/gen/DBusWpaNetwork.c            \

I noticed, we checked in DbusBluez.c source to CHIP, and this file should also get generated in run-time. We need to fix this issue to make BLE and WIFI consistent

Itz-RogueX, 2020-09-21:
I'll do it.

PR #2815 (2020-09-25): [linux] Delete the dbus generated files
Problem
DbusBluez.c is auto-generated in run-time and its source file also get checked in to the repository

Summary of Changes
Remove file DbusBluez.c and use auto generated file

fixes #1936
closed 2020-09-25 by rwalker-apple