project-chip/connectedhomeip #1940: Figure out what to do with emberAfStackDown

bzbarsky-apple, 2020-07-31:
#### Problem
The initial landing in #1609 includes a function called `emberAfStackDown` in `util.c` that none of the other pulled-in code seems to use.  It's called from `af-main-common.c` in the original silabs code.

Various parts of this function (calls to `emberStackIsPerformingRejoin`, `emberNetworkState`, `emberGetCurrentNetwork`) were commented out to make it compile.

 #### Proposed Solution
We should figure out whether something in the CHIP stack should be calling this function.  If so, we need to figure out what that "something" is, and what to do with the commented-out pieces.

@bhaskar-apple @rwalker-apple @selissia @wehale @tecimovic

holbrookt, 2022-02-01:
@bzbarsky-apple does this need to be fixed for 1.0?

bzbarsky-apple, 2022-02-01:
I doubt it, except insofar as this was doing something important in the original silabs code (what?) and we should make sure we are doing that in Matter.

@selissia do you know what this function was supposed to do?

closed 2023-02-24 by andy31415