project-chip/connectedhomeip #1949: Decide whether invalid endpoint should lead to an error response

bzbarsky-apple, 2020-07-31:
#### Problem
In the SiLabs ZCL implementation, and in the DSR one as well, if the endpoint id is not recognized the message is silently dropped on the floor.  There is no "Default Response" (unlike in the "unrecognized cluster") case, and the spec doesn't seem to define an error code that would be used in this situation either.

 #### Proposed Solution
Decide whether for CHIP we want the "silently drop" behavior or some sort of "Default Response" with an error code.

@CamWms @mrjerryjohns @woody-apple @bhaskar-apple @rwalker-apple

vivien-apple, 2020-11-02:
Personally I would at least have expected a `INVALID_FIELD` result. Better would be an `UNSUPPORTED_ENDPOINT`or `INVALID_ENDPOINT` to match the `UNSUPPORTED_CLUSTER` case.

bzbarsky-apple, 2020-11-07:
Fundamentally, this is a spec-level data/interaction model decision.  We will hash it out there, I guess.