project-chip/connectedhomeip #1967: Implement a generic dispatch of requests

andy31415, 2020-08-04:
#### Problem
After #1950, messages may have different protocols. ZCL data is standard however we also need to be able to handle additional types (vendor specific protocols).

 #### Proposed Solution
Add message dispatch, including support for generic 'any' message receiving and/or filtering for specific protocol types.

woody-apple, 2020-08-04:
@andy31415 Is this a request to support clusters, or custom code added to CHIP to write new functionality outside of the CHIP protocol?

andy31415, 2020-08-04:
My understanding is that we now separate between known protocols (ZCL, maybe others for hanshakes) and proprietary ones (things that ZCL does not define - fictive example of 'my device streams 3d environment scan data').

In this contect, my app should be able to setup things like:
   - handle my standard things (zcl)
   - handle my specific custom protocol things using a separate callback (proprietary parser?)
   - handle anything else differently (e.g. log as 'unknown request')

andy31415, 2020-08-04:
The request is to be able to specify the parse callback depending on the 'protocol type' field in the header. ZCL is standard, anything else goes to a custom callback

tcarmelveilleux, 2020-08-04:
@woody-apple Re: `is encryption required?`, since message protocol ID is in the encrypted portion of messages, then I would say that concern does not apply, since this issue concerns dispatch of messages based on protocol ID, which is always within encrypted messages.

kghost, 2020-11-05:
As `ExchangeManage` already merged, I think we can safely close this issue.

closed 2021-02-03 by stale[bot]