project-chip/connectedhomeip #2061: Import ../third_party/connectedhomeip/src/app/clusters/on-off-client

franck-apple, 2020-08-11:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2020-08-11:
@franck-apple I'm not sure what this issue is about.  Where is that path (`../third_party/connectedhomeip/src/app/clusters/on-off-client`) meant to be rooted at?

franck-apple, 2020-08-11:
My mistake. It's actually ../third_party/connectedhomeip/src/app/clusters/on-off (I corrected the title)
Client part needs to be imported.
Maybe @rwalker-apple can bring some clarity.

bzbarsky-apple, 2020-08-11:
@rwalker-apple and I talked about this yesterday a bit.  It looks like there is no client part to on-off in the silabs code.  The only commands for it ("Read Attributes Response" and "Default Response", afaict) are not cluster-specific.

rwalker-apple, 2020-08-11:
yep.  no client.

franck-apple, 2020-08-11:
Got it. Closing the issue then.

closed 2020-08-11 by franck-apple