project-chip/connectedhomeip #2080: Remote tunneling support for WML baseline

andy31415, 2020-08-12:
#### Problem
Change #2077 added an unchanged (except license and renaming) copy of weave message layer into chip.

This copy needs to be cleaned up and unnecessary parts removed.

 #### Proposed Solution
One item to remove is any tunneling support. (all ifdefs and dependencies)

yufengwangca, 2020-08-13:
Currently,  tunneling support exists throughout CHIP stack which include SystemLayer, InetLayer, PlatformLayer (DeviceConfig, ConnectivityManager), MessageLayer, Configuration etc..

Is this issue cover message layer tunnel support only or all the tunnel support in CHIP stack?

andy31415, 2020-08-13:
I believe we should remove it throughout. Tunneling is currently not supported by CHIP spec.

PR #2105 (2020-08-14): Remove tunneling support from CHIP code base
Problem
Tunneling is a feature inherited from Weave and this feature will not be used in CHIP.

Proposed Solution
Cleanup all tunneling support from CHIP code base.

fixes #2080
closed 2020-08-14 by BroderickCarlin