project-chip/connectedhomeip #2084: on/off broken on nrf5 lock app

mspang, 2020-08-12:
As of #2014, sending on/off commands to nRF5 lock app via Thread appears to crash the device

After:
[00000000] <info> app: Packet received from UDP:FE80::95E8:5D28:AAF8:B330:11095: u bytes                                       
[00000000] <info> app: APS frame processing success!                                                                           
[00000000] <info> chip: [ZCL] RX len 3, ep 1, clus 0x 6                                                                        
[00000000] <info> chip: [ZCL]  FC 1 seq 1 cmd 0 payload[                                                                       
[00000000] <info> chip: [ZCL] NULL                                                                                             
[00000000] <info> chip: [ZCL] ]      

<device unresponsive>

Before:
[00000000] <info> chip: [IN] New peer connection received.
[00000000] <info> app: Received a new connection.
[00000000] <info> app: Packet received from UDP:FE80::95E8:5D28:AAF8:B330:11095: u bytes
[00000000] <debug> chip: [ZCL] RX: On
[00000000] <info> app: Data model processing success!

mspang, 2020-08-12:
@bzbarsky-apple

bzbarsky-apple, 2020-08-12:
Unfortunately I don't have a good way to test the nrf5 lock app locally...  I assume it uses `examples/platform/nrf528xx/app/Server.cpp` and we do reach the `InitDataModelHandler` call there?

After that worth checking whether we reach the tops of the following functions before crashing:

1. `emberAfOnOffClusterOnCallback` in `src/app/clusters/on-off-server/on-off.c`
1. `emberAfSendDefaultResponseWithCallback` in `src/app/util/util.c`

Knowing that would help pin down where things go off the rails.

bzbarsky-apple, 2020-08-12:
Also worth checking whether `emberAfPostAttributeChangeCallback` in `examples/lock-app/nrf5/main/DataModelHandler.cpp` is getting called.

mspang, 2020-08-12:
Thanks for those pointers, let me take a look.

mspang, 2020-08-12:
It seems to get through all that. Is NULL expected in the below output? Let me try the debugger..

[00000000] <info> app: Received a new connection.
[00000000] <info> app: Packet received from UDP:FE80::95E8:5D28:AAF8:B330:11095: u bytes                                                                                
[00000000] <info> app: APS frame processing success!            
[00000000] <info> chip: [ZCL] RX len 3, ep 1, clus 0x 6         
[00000000] <info> chip: [ZCL]  FC 1 seq 1 cmd 1 payload[        
[00000000] <info> chip: [ZCL] NULL                                                  
[00000000] <info> chip: [ZCL] ]                                                     [00000000] <info> app: emberAfOnOffClusterSetValueCallback...
[00000000] <info> app: emberAfPostAttributeChangeCallback       
[00000000] <info> app: emberAfSendResponseWithCallback...

bzbarsky-apple, 2020-08-12:
>  Is NULL expected in the below output?

Yes, it is: this particular command has no command arguments.

Does the logging there say we land in `emberAfSendResponseWithCallback`, not just `emberAfSendDefaultResponseWithCallback`?  Do we then land in `chipSendResponse` in `src/app/util/chip-response.cpp`?  

Is there a sane `ChipResponseDestination` at that point, if we do?

mspang, 2020-08-12:
It's faulting in printf().. let me try removing that.

<HardFault Exception>
--
uxListRemove
xTaskRemoveFromEventList
xQueueSemaphoreTake
xQueueTakeMutexRecursive
__retarget_lock_acquire_recursive
__sinit
printf
doEncodeApsFrame
encodeApsFrame
chipSendResponse
emberAfSendResponseWithCallback
emberAfSendDefaultResponseWithCallback
emberAfSendImmediateDefaultResponseWithCallback
emberAfSendImmediateDefaultResponse
emberAfOnOffClusterOffCallback
emberAfOnOffClusterServerCommandParse
emberAfClusterSpecificCommandParse
emAfProcessClusterSpecificCommand
dispatchZclMessage
emberAfProcessMessage
ServerCallback::HandleDataModelMessage
ServerCallback::OnMessageReceived
SecureSessionMgrBase::HandleDataReceived
Tuple<chip::Transport::UDP>::OnMessageReceive
UDP::HandleMessageReceived
UDP::OnUdpReceive
IPEndPointBasis::HandleDataReceived
UDPEndPoint::HandleDataReceived
InetLayer::HandleInetLayerEvent
Layer::HandleEvent
GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToSystemLayer
GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent
GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::DispatchEvent
GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop
GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::RunEventLoop
GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain
vPortEnableVFP
Top of stack - invalid return address location:

mspang, 2020-08-12:
Removing printf fixes the fault.

bzbarsky-apple, 2020-08-13:
Sounds like we should convert that stuff to using either the ember-print bits in `src/app/util` (which need a better name) or to `chip::Logging` more directly.  I can work on that; thank you for hunting this down!

mspang, 2020-08-13:
printf as currently implemented should cause a no-op write rather than crash the device so I spent some time investigating that.

I've concluded that the newlib retargeted locking primitives are incorrectly implemented (invalid typecasts) and are corrupting memory. Implementing them correctly on top of FreeRTOS semaphores actually looks rather awkward. I'm looking into whether we should fix the newlib locking or if we could instead just remove it.

bzbarsky-apple, 2020-08-13:
@mspang It sounds like https://github.com/project-chip/connectedhomeip/pull/2100 will fix the crash and we should mutate this issue to be about the missing output due to the use of `printf`, right?

@bhaskar-apple Do you have time to work on switching to `chip::Logging`, directly or indirectly, here?

mspang, 2020-08-13:
Yes, #2100 will fix crash and logging can be done as a followup. It's no big deal to have the printfs after my fix, but note that they do not currently get printed.

mspang, 2020-08-13:
Reopened for the followup logging cleanup.

bzbarsky-apple, 2020-10-01:
This got fixed by <https://github.com/project-chip/connectedhomeip/pull/2103>: we now use CHIP logging.

PR #2100 (2020-08-13): [nrf5] Repair the newlib locking
Calling stdio functions in certain circumstances currently crashes the
device. Enabling configUSE_LIST_DATA_INTEGRITY_CHECK_BYTES makes this
more likely to happen.

The problem is that the types used in FreeRTOSNewlibLockSupport.c don't
match those used by newlib; we're mixing up _LOCK_T (struct __lock *) with
its content.

Fix the implementation and remove all the typecasts. Note that we need a
2nd allocation for the dynamic case as FreeRTOS and newlib APIs don't
match up well.

Possibly we should remove this code, but for now just repair it.

fixes #2084
commit c1ff7f9e (2020-08-13): [nrf5] Repair the newlib locking (#2100)

* [nrf5] Repair the newlib locking

Calling stdio functions in certain circumstances currently crashes the
device. Enabling configUSE_LIST_DATA_INTEGRITY_CHECK_BYTES makes this
more likely to happen.

The problem is that the types used in FreeRTOSNewlibLockSupport.c don't
match those used by newlib; we're mixing up struct __lock * with its
content.

Fix the implementation and remove all the typecasts. Note that we need a
2nd allocation for the dynamic case as FreeRTOS and newlib APIs don't
match up well.

Possibly we should remove this code, but for no [... cut]
commit bcb7b5c8 (2020-08-13): GN: Fix the Mac build (#2102)

* GN: Fix the Mac build

* Enable more debug printing in Silicon Labs code (#2087)

* [nrf5] Repair the newlib locking (#2100)

* [nrf5] Repair the newlib locking

Calling stdio functions in certain circumstances currently crashes the
device. Enabling configUSE_LIST_DATA_INTEGRITY_CHECK_BYTES makes this
more likely to happen.

The problem is that the types used in FreeRTOSNewlibLockSupport.c don't
match those used by newlib; we're mixing up struct __lock * with its
content.

Fix the implementation and remove all the typecasts. Note that we need a
2nd allocation f [... cut]
closed 2020-10-01 by bzbarsky-apple