project-chip/connectedhomeip #2093: Migrate WML crypto usage ot CHIP /crypto

andy31415, 2020-08-13:
#### Problem
For compile purposes, WML baseline added headers-only of original weave crypto. 

 #### Proposed Solution
These should be replaced with /crypto includes and long term goal is to align crypto support with Secure Channel Tiger Team crypto instructions. 

Only crypto defined for CHIP Spec should be used instead original weave core constructs.

yufengwangca, 2020-08-15:
We don't need to migrate WML crypto to CHIP crypto since all users of current WML crypto are going to retire and replaced with SCTT defined new protocol.

We need to take this as sub-task of #2096. Close this one for now.

yufengwangca, 2020-08-20:
Besides SecurityManager and profile, WML crypto is stilled used by other components. Reopen this issue.

yufengwangca, 2020-11-20:
Not valid since WML solution is not selected

closed 2020-11-20 by yufengwangca