project-chip/connectedhomeip #2096: Use SCTT defined session establishment protocol within WML

andy31415, 2020-08-13:
#### Problem
Currently WML baseline was ported with headers-only for dependencies, one of which is the 'security profile'. 

The security profile is supposed to handle session creation, however it uses the weave defined methodology.

 #### Proposed Solution
Replate weave session creation with SCTT defined protocol. This may include removing 'security profile' and replacing with a new one (if protocols are a rewrite) or implement 'security profile' as as the CHIP protocol instead of bring in weave code (assuming interfaces are compatible enough to warrant reuse).

Important to note that SCTT defined protocol requires some additional reliability guarantess for the handshake and it currently only defines TCP as meeting these needs. Need to determine if TCP is required or if WRM can be used and how to encapsulate these requirements within WML baseline.

yufengwangca, 2020-08-15:
I would like to use this ticket to focus on SecurityManager and Security Profile refactor. I found this task is closely coupled with https://github.com/project-chip/connectedhomeip/issues/2093 which I am currently working on.

We need to handle them together instead of individually.

yufengwangca, 2020-08-19:
This task is not complete yet.

PR #2209 (2020-08-18): Replace CHIP Security Monitor Memory Manager within CHIPSecuityManage…
This is part of SecurityManager cleanup which replace CHIP Security Monitor Memory Manager within CHIPSecuityManager with the on-system security memory manager. 

Those touched logic will get further cleaned up as we migrate SecurityMamanger to  SCTT defined security protocol. And this change is just to get current Security Manager linked.

fixes #2096