project-chip/connectedhomeip #2216: Remove or define CHIP_CONFIG_ENABLE_EPHEMERAL_UDP_PORT

woody-apple, 2020-08-18:
Similar to above, what feature is this for?

_Originally posted by @woody-apple in https://github.com/project-chip/connectedhomeip/pull/2206_

yufengwangca, 2020-08-20:
@woody-apple 

This is for enabling the use of an ephemeral src port for the UDP endpoint instead of the fixed CHIP port, this would allow multiple exchanges to be made across processes using UDPEndPoints.

In Weave, we have both Nest App and the Google Home App, we need to allow them to run simultaneously. This feature allows multiple simultaneous Weave initiating exchanges over UDP can be made by client apps and all of them would have a different Weave src port.

I think we might need this feature for CHIP at some point, not not likely in CHIP v1.0. Let's me know if you prefer to clean this up at this moment.

msandstedt, 2020-10-08:
This is absolutely needed for CHIP.

CHIP clients that wish to talk to multiple server devices should be listening on ephemeral ports.  Right now, if you try to instantiate multiple controllers within a client, they all bind to port 11095 which breaks a lot of things.

This workaround is, in part, allowing us to run multiple controller instances at the moment:

```
diff --git a/src/transport/UDP.h b/src/transport/UDP.h
index f2287588..6e7a923a 100644
--- a/src/transport/UDP.h
+++ b/src/transport/UDP.h
@@ -74,7 +74,7 @@ public:
 private:
     Inet::InetLayer * mLayer         = nullptr;               ///< Associated inet layer
     Inet::IPAddressType mAddressType = kIPAddressType_IPv6;   ///< type of listening socket
-    uint16_t mListenPort             = CHIP_PORT;             ///< UDP listen port
+    uint16_t mListenPort             = 0;             ///< UDP listen port
     InterfaceId mInterfaceId         = INET_NULL_INTERFACEID; ///< Interface to listen on
 };
```

yufengwangca, 2021-01-07:
This cleanup work is not needed since whole WML based messaging stack has been removed.

closed 2021-01-07 by yufengwangca