project-chip/connectedhomeip #2220: MbedTLS add_entropy_source is broken with no default source

mspang, 2020-08-18:
This function only works if you already have an entropy source.

```
CHIP_ERROR add_entropy_source(entropy_source fn_source, void * p_source, size_t threshold)
{
    CHIP_ERROR error              = CHIP_NO_ERROR;
    int result                    = 0;
    EntropyContext * entropy_ctxt = NULL;

    VerifyOrExit(fn_source != NULL, error = CHIP_ERROR_INVALID_ARGUMENT);

    **entropy_ctxt = get_entropy_context();**
    VerifyOrExit(entropy_ctxt != NULL, error = CHIP_ERROR_INTERNAL);

    result = mbedtls_entropy_add_source(&entropy_ctxt->entropy, fn_source, p_source, threshold, MBEDTLS_ENTROPY_SOURCE_STRONG);
    VerifyOrExit(result == 0, error = CHIP_ERROR_INTERNAL);
exit:
    return error;
}
```

Bold line fails without an initial source because it calls `mbedtls_ctr_drbg_seed`.

pan-apple, 2020-08-19:
@mspang, how do I reproduce this error? Does it show up if crypto tests are run against a certain platform build? I think I have a fix, but want to reproduce the failure and test the fix.

mspang, 2020-08-19:
I found this with the EFR32 bringup.

To repro:
  - Set MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES in the MbedTLS config file header remove the default sources
  - Try to add an app specific entropy source via the API. This fails.