project-chip/connectedhomeip #2238: Need 'make pretty' replacement

woody-apple, 2020-08-19:
#### Problem
Make pretty doesn't work in GN/Ninja land.

 #### Proposed Solution
We should move over to a local restyle

mspang, 2020-08-19:
This seems broken today. Marking non-blocking

gerickson, 2020-10-05:
+1, just tried a PR today. Life seems much harder on this front than previously.

mspang, 2020-10-06:
Did you see restyle-diff.sh ? Would be worth knowing what's missing from that to close this out.

woody-apple, 2020-10-06:
@rwalker-apple this is done right? Can you confirm it's a drop in replacement? (I think it is!)

rwalker-apple, 2020-10-07:
`restyle-diff.sh` aims to replicate what restyled.io does in CI.  it pulls down the restyled.io CLI and runs it on any file with diffs from a supplied refspec (as the first argument, which defaults to `master`)

closed 2020-10-07 by rwalker-apple