project-chip/connectedhomeip #2243: Secure Pairing must run on a reliable messaging protocol

pan-apple, 2020-08-19:
#### Problem
The pairing protocol is currently using unreliable messaging (e.g. BLE). The packets could be lost during transmission. Need a mechanism to detect and retransmit dropped packets.

 #### Proposed Solution
The reliable messaging will need acknowledgement (from peer) of transmitted packets, timeouts, and retransmissions (of dropped packets). 

This could be handled inside the secure pairing protocol, as it already performs a handshake between the peers. The handshake involves crypto operations on the low powered devices, that adds delays in the handshake.

The acknowledgement of the received packets can be faster than the pairing handshake (i.e. ack as soon as the message is received). This will help with shorter timeouts, and faster retries for the dropped packets.

So, it'll be better if a separate layer could guarantee the reliable messaging.

RyanTheOptimist, 2020-08-19:
Sounds great! It seems like we should be able to use (mostly) the same machinery for sending this messages reliably that we use when sending operational messages. (ACKs, retransmissions, etc)

woody-apple, 2021-01-25:
SW BRB: @pan-apple @RyanTheOptimist what is the current POR here? Is this still needed post CRMP?

pan-apple, 2021-01-27:
This is not an issue for the following use-cases
1. PASE pairing over BLE
2. CASE pairing (as it uses TCP)

This can be an issue for PASE pairing over SoftAP and ethernet. As it stands today, `CRMP` works on top of Secure Session, i.e. it cannot be used for unencrypted traffic exchange. PASE session handshake involves sending unencrypted messages. So `CRMP` cannot be used to solve it.

woody-apple, 2021-04-05:
Assigning to @pan-apple

PR #5938 (2021-04-14): Use exchange for session establishment and provisioning
 #### Problem
Session establishment doesn't use exchange or reliable messaging.

 #### Summary of Changes
- Add concept of ExchangeTransport to distinguish between encrypted
  and unencrypted message exchanges
- Update PASE to use exchange manager to send/receive messages
- Update network provisioning code to use exchange manager

 Fixes #2240, Fixes #2243
closed 2021-04-14 by yufengwangca