project-chip/connectedhomeip #2248: Implement SCTT defined session establishment protocol in SecurityManager

yufengwangca, 2020-08-19:
Problem
Currently WML baseline was ported with headers-only for dependencies, one of which is the 'security profile'.

The security profile is supposed to handle session creation, however it uses the weave defined methodology.

Proposed Solution
Replate weave session creation with SCTT defined protocol. This may include removing 'security profile' and replacing with a new one (if protocols are a rewrite) or implement 'security profile' as as the CHIP protocol instead of bring in weave code (assuming interfaces are compatible enough to warrant reuse).

Important to note that SCTT defined protocol requires some additional reliability guarantess for the handshake and it currently only defines TCP as meeting these needs. Need to determine if TCP is required or if WRM can be used and how to encapsulate these requirements within WML baseline

woody-apple, 2020-08-19:
@yufengwangca How is this different than the currently implemented security? (which fits SCFTT)

yufengwangca, 2020-08-19:
@woody-apple 

The goal of this ticket is to run current implemented security protocol on top of WML APIs, that means we either need to update the existing implementation within transport layer or plug in the current security protocol implementation to SecurityManager class. 

The details of this task needs further discussion and agreement.

closed 2021-02-03 by stale[bot]