project-chip/connectedhomeip #2250: GN build parity - NXP

mspang, 2020-08-19:
NXP is landing:
https://github.com/project-chip/connectedhomeip/pull/2234

Add it to the GN build.

mspang, 2020-09-08:
@doru91 How disruptive would it be to remove the automake support and add NXP support as a follow-on?

doru91, 2020-09-09:
@mspang , I have to do an analysis first as I'm not familiar with gn/ninja. I'll privide an accurate estimate next week (when I return from vacation).

doru91, 2020-09-28:
@mspang , @woody-apple I am currently working on adding GN/Ninja support on NXP platform. Hope it will be ready by the end of the week.

mspang, 2020-09-28:
Please LMK if you need any support. I have a NXP dev kit on my desk now.

doru91, 2020-10-01:
@mspang , I would be really grateful if you can help me. In the last day, I've been struggling with a dependancy cycle error:

```
doru@ubuntu:~/Desktop/doru91/connectedhomeip/examples/lock-app/k32w$ gn gen out/debug --args="k32w_sdk_root=\"${K32W061_SDK_ROOT}\""
lock_app.executable
lock_app.executable
ERROR Dependency cycle:
  //third_party/connectedhomeip/examples/lock-app/k32w:k32w ->
  //third_party/connectedhomeip/examples/lock-app/k32w:lock_app ->
  //third_party/connectedhomeip/examples/lock-app/k32w:lock_app.flashing ->
  //third_party/connectedhomeip/examples/lock-app/k32w:lock_app.image ->
  //third_party/connectedhomeip/examples/lock-app/k32w:lock_app.executable ->
  //third_party/connectedhomeip/examples/lock-app/k32w:k32w
```

Steps to reproduce:

1. clone nxp_smart_lock_demo_app_v5 branch: https://github.com/doru91/connectedhomeip/tree/nxp_smart_lock_demo_app_v5
2.  git submodule update --init
3. source third_party/connectedhomeip/scripts/activate.sh
4. export K32W061_SDK_ROOT=/home/doru/Desktop/SDK_K32_2.6.1 (SDK can be downloaded by followinf the instructions from https://github.com/doru91/connectedhomeip/tree/nxp_smart_lock_demo_app_v5/examples/lock-app/k32w but for reproducing this problem an empty folder can be used)
5. gn gen out/debug --args="k32w_sdk_root=\"${K32W061_SDK_ROOT}\""

mspang, 2020-10-02:
Remove this line:

```
diff --git a/examples/lock-app/k32w/BUILD.gn b/examples/lock-app/k32w/BUILD.gn
index 02c0f8e8..b56d3dad 100644
--- a/examples/lock-app/k32w/BUILD.gn
+++ b/examples/lock-app/k32w/BUILD.gn
@@ -66,7 +66,6 @@ k32w_executable("lock_app") {
     ":sdk",
     "${chip_root}/examples/common/chip-app-server:chip-app-server",
     "${chip_root}/examples/common/chip-app-server:chip-app-server",
-    "${chip_root}/examples/lock-app/k32w",
     "${chip_root}/src/lib",
     "${k32w_platform_dir}/app/support:freertos_mbedtls_lock_support",
     "${openthread_root}:libopenthread-cli-ftd",

```

doru91, 2020-10-02:
Thanks a lot, Michael. Finally, I am able to start solving compilation issues.

doru91, 2020-10-05:
@mspang , I managed to build the project using GN/Ninja. In order to build external mbedtls both for OT and Project CHIP I had to do some hacks so I need some help before sending the final PR:

1. https://github.com/doru91/connectedhomeip/commit/480b73dfa36fa532f7dd698bbd38e23a4b1ed2dd - here I add basic GN/Ninja support and the changes look good to me;
2. https://github.com/doru91/connectedhomeip/commit/f3c8399a45f1bf5b6dbba378fcc6e288e3e1ed16 - here I had to do some ugly changes to the BUILD.gn file inside third_party/mbedtls/BUILD.gn. Basically I wanted to transmit the includes and defines from third_party/k32w_sdk/k32w_sdk.gni to third_party/mbedtls/BUILD.gn and I ended up by copying those includes/defines in both files. Could you please advise on how I can do this more elegant?

And one more thing: how could I override the -Og flag such that is used only for the ELock application/platform files while -Os is used for all the other builds (OT, mbedtls)?

doru91, 2020-10-06:
Even with is_debug=false (-Os) the generated image is very large:

```
doru@ubuntu:~/Desktop/demo/connectedhomeip/examples/lock-app/k32w$ gn gen out/debug --args="k32w_sdk_root=\"${K32W061_SDK_ROOT}\" is_debug=false"
Done. Made 94 targets from 81 files in 31ms
doru@ubuntu:~/Desktop/demo/connectedhomeip/examples/lock-app/k32w$ ninja -C out/debug/
ninja: Entering directory `out/debug/'
[1/1] Regenerating ninja files
[7/7] stamp obj/lock_app.image.stamp
doru@ubuntu:~/Desktop/demo/connectedhomeip/examples/lock-app/k32w$ ls -l out/debug/
total 24528
-rw-rw-r-- 1 doru doru       69 Oct  6 01:39 args.gn
-rw-rw-r-- 1 doru doru    32537 Oct  6 01:39 build.ninja
-rw-rw-r-- 1 doru doru     4988 Oct  6 01:39 build.ninja.d
-rw-rw-r-- 1 doru doru   213882 Oct  6 01:16 chip-app-server.a
-rwxrwxr-x 1 doru doru 21807232 Oct  6 01:39 chip-k32w061-lock-example
-rwxr-xr-x 1 doru doru      232 Oct  5 23:33 chip-k32w061-lock-example.flash.py
-rwxrwxr-x 1 doru doru   934534 Oct  6 01:39 chip-k32w061-lock-example.hex
-rw-rw-r-- 1 doru doru  2126454 Oct  6 01:39 chip-k32w061-lock-example.map
-rw-rw-r-- 3 doru doru     9845 Oct  1 10:54 firmware_utils.py
drwxrwxr-x 4 doru doru     4096 Oct  5 23:33 gen
-rwxrwxr-x 2 doru doru     4263 Oct  1 23:34 k32w_firmware_utils.py
drwxrwxr-x 2 doru doru     4096 Oct  6 01:39 lib
drwx------ 5 doru doru     4096 Oct  5 23:33 obj
-rw-rw-r-- 1 doru doru    28125 Oct  6 01:39 toolchain.ninja
doru@ubuntu:~/Desktop/demo/connectedhomeip/examples/lock-app/k32w$ 
```

doru91, 2020-10-08:
@mspang , kind reminder.

mspang, 2020-10-09:
My apologies for the delay.

Can you try incorporating the change in this branch:
https://github.com/mspang/connectedhomeip/tree/k32w-test

Here's an explanation of how config propagation works:

```
gn help configs

Ordering of flags and values

  1. Those set on the current target (not in a config).
  2. Those set on the "configs" on the target in order that the 
     configs appear in the list.
  3. Those set on the "all_dependent_configs" on the target in order
     that the configs appear in the list.
  4. Those set on the "public_configs" on the target in order that
     those configs appear in the list.
  5. all_dependent_configs pulled from dependencies, in the order of
     the "deps" list. This is done recursively. If a config appears
     more than once, only the first occurrence will be used.
  6. public_configs pulled from dependencies, in the order of the
     "deps" list. If a dependency is public, they will be applied
     recursively.
```

The key one here is `public_configs pulled from dependencies`. If we add a dependency on the SDK to the mbedtls target, we'll get the options applied. If we make that dependency public, it will also apply to targets that depend on mbedtls (which is needed if the configs affect header evaluation).

The trick is doing it without introducing cycle and we have to bend over backwards for that in several places (e.g. thread configuration influencing mbedtls configuration requires information to flow backwards; openthread depends on mbedtls, but mbedtls config depend on openthread config. This reversal is implemented by adding header-only library targets that propagate the include paths and defines).

To add the K32W bits to the mbedtls build file, we can edit the build file with platform-specific conditionals, add some build arguments and specify them in args.gni, or make a platform specific copy of the mbedtls target. My branch above implements the last one. Other platforms have their own copy of the entire MbedTLS library and this approach is close to that.

doru91, 2020-10-13:
I am grateful for your help @mspang. I believe this issue can be closed now as the PR adding GN was merged to master.

closed 2020-10-14 by mspang