project-chip/connectedhomeip #2375: GN build fails on Linux.

eliotstock, 2020-08-30:
#### Problem

The GN build is broken. I was on commit 0700e4f7588055399912e8606d38fe502df6d1a3. I followed the instructions in docs/BUILDING-GN.md.

```
$ gn gen out/host
ERROR at //build/config/linux/pkg_config.gni:97:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /home/eliotstock/connectedhomeip/out/host/
Command: python /home/eliotstock/connectedhomeip/gn/build/config/linux/pkg-config.py dbus-1
Returned 1.
stderr:

Package dbus-1 was not found in the pkg-config search path.
Perhaps you should add the directory containing `dbus-1.pc'
to the PKG_CONFIG_PATH environment variable
No package 'dbus-1' found
Could not run pkg-config.

See //third_party/ot-br-posix/BUILD.gn:25:1: whence it was called.
pkg_config("dbus_config") {
^--------------------------
See //src/platform/BUILD.gn:229:26: which caused the file to be included.
        public_deps += [ "${ot_br_posix_root}:ot_br_client" ]
                         ^---------------------------------
```

mspang, 2020-08-30:
sudo apt-get install libdbus-1-dev

eliotstock, 2020-08-30:
Thanks, that fixed that one, but it seems I'm going to need loads more:

```
$gn gen out/host
ERROR at //build/config/linux/pkg_config.gni:97:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /home/e/r/connectedhomeip/out/host/
Command: python /home/e/r/connectedhomeip/gn/build/config/linux/pkg-config.py glib-2.0 gio-unix-2.0
Returned 1.
stderr:

Package glib-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `glib-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'glib-2.0' found
Package gio-unix-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gio-unix-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gio-unix-2.0' found
Could not run pkg-config.

See //gn/chip/linux/BUILD.gn:17:1: whence it was called.
pkg_config("glib") {
^-------------------
See //gn/chip/linux/gdbus_library.gni:88:7: which caused the file to be included.
      glib_config,
      ^----------

```

Installing glib-2.0 didn't solve the first error. Is there a reliable way to determine the Debian/Ubuntu package I need based on the "No package..." message?

mspang, 2020-08-31:
libglib2.0-dev

https://packages.debian.org/search?searchon=contents&keywords=%2Fglib-2.0.pc&mode=path&suite=stable&arch=any

PR #2578 (2020-09-11): Add libdbus-1-dev & libglib2.0-dev to build dependencies
We've added some build dependencies that need to be documented. Add
them.

Closes #2375
commit 774bdd5b (2020-09-11): Add libdbus-1-dev & libglib2.0-dev to build dependencies (#2578)

We've added some build dependencies that need to be documented. Add
them.

Closes #2375
closed 2020-09-11 by mspang