project-chip/connectedhomeip #2390: Need a second endpoint for the wifi-echo server

bzbarsky-apple, 2020-08-31:
#### Problem
We want to demo multiple endpoints at some point.  Right now the wifi-echo server is configured with only one endpoint.

We used to have two endpoints, with the on/off cluster on both of them, before, when we had the dotdot generated code.

 #### Proposed Solution
Add second endpoint, maybe with just the on-off cluster on it at the moment.

I'm sorry I didn't catch this earlier...  Looking at how to do this most easily.

@rwalker-apple

rwalker-apple, 2020-08-31:
> #### Problem
> We want to demo multiple endpoints at some point. Right now the wifi-echo server is configured with only one endpoint.
>
> We used to have two endpoints, with the on/off cluster on both of [...]

I think the cleanest thing for echo would be to move it to another protocol id.  is protocol id available in the message layer yet?

bzbarsky-apple, 2020-09-01:
Ah, sorry, I wasn't very clear.  This is changes to the "echo" app, but have nothing to do with the echo functionality.  This is about background work needed to fix https://github.com/project-chip/connectedhomeip/issues/1096 -- have independent on/off control for two different lights.

closed 2020-09-01 by mspang