project-chip/connectedhomeip #2412: CHIP's ZCL cluster implementations have assumed sizes for attributes in the store

rwalker-apple, 2020-09-02:
#### Problem
Code that reads and writes attributes in response to a command in cluster implementations has hard-coded size information.  These sizes and types should come from code generated from the spec.

 #### Proposed Solution
-[ ] Generate code that binds attributes to their sizes from schema "truth"
-[ ] Consume those generated defines, or whatever in the read/write attribute code in cluster implementations

rwalker-apple, 2020-09-02:
@bzbarsky-apple this is a biggie ;)

bzbarsky-apple, 2021-03-03:
@vivien-apple Could we spit out `#define`s somewhere for this?  I guess one problem is that it's the SDK that would need to consume these `#define`s in many cases, so it might be generated code but checked into the SDK itself....