project-chip/connectedhomeip #2435: Sort out APIs for message sending for data model

bzbarsky-apple, 2020-09-02:
#### Problem
The Silicon Labs code has a bunch of APIs for message sending.  These can do different things depending on the `EmberOutgoingMessageType` passed in, have some sort of "sent" callbacks, etc.

 #### Proposed Solution
1) Figure out what values of `EmberOutgoingMessageType` we actually need to support on the API level (as opposed to down in the CHIP library guts).
2) Figure out how to support them.
3) Figure out whether we need the callback bits.  @wehale @selissia @tecimovic what are those used for?
4) Figure out whether we need a `ChipResponseDestination` passed through all this stuff or whether we could nix that struct and just pass around `chip::NodeId` plus some globals for getting the secure channel bits.

@pan-apple do you have any bandwidth to think about this stuff?

@bhaskar-apple @shana-apple @rwalker-apple

bzbarsky-apple, 2020-09-02:
One specific issue which causes this to block <https://github.com/project-chip/connectedhomeip/issues/2057>: `iasZoneClientServiceDiscoveryCallback` is handed an `EmberAfServiceDiscoveryResult`.  It then tries to send a command that that results node id.  We can probably rejigger this stuff to store a `chip::NodeId` but making it store a `ChipResponseDestination` does not seem too plausible.

pan-apple, 2020-09-02:
@bzbarsky-apple , I can take a look at it. I'll reach out to you on IM to discuss the use-case further.