project-chip/connectedhomeip #2470: Implement stubs in zcl reporting cluster

bhaskar-apple, 2020-09-04:
#### Problem
Implement stubs in zcl reporting cluster e.g.
```
uint32_t halCommonGetInt32uMillisecondTick(void)
{
    return 0; // Stub for now.
}

EmberStatus emberAfEventControlSetDelayMS(EmberEventControl * control, uint32_t delayMs)
{
    return EMBER_SUCCESS; // Stub for now
}
```

 #### Proposed Solution
Implement it

bzbarsky-apple, 2020-10-15:
@rwalker-apple @bhaskar-apple Until this is done, most of the code in reporting.c gets "optimized out" by the compiler because it detects undefined behavior and makes some assumptions about some early-continue `if` statements always testing true.  This is also pointed out by the "uninitialized variable" warnings that are manually suppressed.....

vivien-apple, 2020-11-02:
`halCommongGetInt32uMillisecondsTick` has been done by #3553 

The token specific bits are tracked by #3275 . I would call this issue close so we can focus on the specific `halCommonGetIndexedToken`and `halCommonSetIndexedToken` bits in #3275

closed 2020-11-02 by vivien-apple