project-chip/connectedhomeip #2490: SPAKE2+ and patents

Emill, 2020-09-07:
Hi. I'm not sure if this is the correct place to ask this, but I noticed that this repo includes code for the SPAKE2+ crypto algorithm for password authenticated key exchange. SPAKE2 is however possibly affected by a patent held by Nokia according to https://datatracker.ietf.org/ipr/4018/, so I was just wondering if you were aware of this and in that case, if Nokia has granted a license to use this algorithm in CHIP?

woody-apple, 2020-10-12:
The declaration referenced at https://datatracker.ietf.org/ipr/4018/ is directed at SPAKE2, not SPAKE2+, and it was not submitted by the patent owner.

closed 2020-10-12 by woody-apple

Emill, 2020-10-12:
Well whether you use SPAKE2+ or SPAKE2 doesn't matter in this case because SPAKE2+ is a superset of SPAKE2. SPAKE2+ does everything that SPAKE2 does plus an extra augmentation step, and is hence affected by the same patent.

And I'm not sure why it matters to you who sent the declaration?